Sloeber/arduino-eclipse-plugin #491: compiling for ATmega2560

belebog, 2016-06-30:
Eclipse Version: Mars.1 Release (4.5.1)
Arduino Eclipse Extensions	2.5.0.201601060217	

Having selected the Mega Board with ATmega 2560 cpu, <avr/io.h> doesn't include the correct "io*.h" file (which should be <avr/iom2560.h>). As an effect, various preprocessor defines fail to be defined (eg. DDRx, PORTx, PINx for Ports E to L)

As a workaround, the define "__AVR_ATmega2560__" can be added to the compiler options.

The same problem affects the code parser in the editor. I have not found a workaround for this yet.

jantje, 2016-06-30:
I just checked in my workspace and avr/iom2560.h is correctly included.
If you look in project properties->C/C++ General->preprocessor include paths...->Enties->GNU C (or GNU C++) open up Arduino compiler settings you should find __AVR_ATmega2560__

As to the work around: If you add the define in the project properties->arduino->add to c++ command line, then codan must do the same as the compiler. In other words the gui should match the compilation.
As that does not seem to be the case at your system I think something else is wrong.

Note that the atmega 2560 is the board I use most often so a problem like this seems pretty unlikely to me.

jantje, 2016-08-20:
@belebog have you some more input? Can this be closed?

jantje, 2016-09-03:
Closing as usage error due to lack of input.

closed 2016-09-03 by jantje