Sloeber/arduino-eclipse-plugin #492: Plugin needs write access to /usr/lib/eclipse/arduinoPlugin

broddo, 2016-06-30:
I installed the plugin as root - which worked fine. But when I run it as a normal user, I'm getting:
"Plugin needs write acces to /usr/lib/eclipse/arduinoPlugin"

I installed Eclipse (CDT) using a package manager. I'd rather not grant write permissions so I'm wondering if there is a way around this? Does the plugin absolutely need write access to that folder?

Platform: Arch Linux
Eclipse: Mars.2 Release 4.5.2
Arduino Eclipse Extensions: 3.0.0.201603312133

jantje, 2016-07-01:
The location is the default location. It can be overruled as discussed with the patrons here https://www.patreon.com/posts/where-do-you-to-3752554

jantje, 2016-08-20:
@broddo can this issue be closed?

broddo, 2016-08-20:
Yeah, absolutely. The plugin works fine now that I granted write
permissions to the plugin folder. I thought I'd raise it as an issue in
case you were unaware that the plugin has this behaviour on linux.

Thanks for all the awesome work!

closed 2016-08-20 by jantje

mjwx4p, 2017-07-07:
jantje your comment isn't a answer. the only thing you done is to link to another side where i have to pay for. very bad from you

jantje, 2017-07-07:
> jantje your comment isn't a answer. the only thing you done is to link to another side where i have to pay for. very bad from you

@mjwx4p and all other people thinking this way
My comment is an answer and it is even accepted by the poster as an answer.
The link to the post is a workaround that is only visible to my patrons as is implied by the comment with the link.  
Just like you I need to eat and currently this plugin is my only income. I have written the code and shared it with the world for free. Is it really so bad to spend 1$(exclusive tax) to get access to some explanations?  
Is it really necessary to say "it is very bad" from me because I want to eat something as well?   
Thanks to the patrons I can extend doing this but with the current support I will have to quit doing this and then you will be left with nothing. Is that what you want?
At this very right day about 9600 builds are done with sloeber a day. Only 87 of the people doing these builds are patrons (thank you patrons) with a total of 238$ a month. And there are also 3 "active" contributors.
How would you feel if someone who is using years of work from you for free and calls you "very bad" because you only share some info with these 87 people who do a small contribution to supporting you?   
  
No thanks.  
Jantje

tenten8401, 2018-06-10:
Is it possible this could be reconsidered, or perhaps an option added to change the write directory of plugin files?

I'm over here on NixOS, and because of the way packaging works, every time there's an update for Eclipse or a dependency eclipse depends on, I'd need to adjust the write permissions again.

It'd be really useful to be able to have the plugin write elsewhere.

jantje, 2018-06-10:
Nearly all OS'es support linking at the folder level. I don't know NixOS but assuming it is a linux you should be able to move the arduinoPlugin folder anywhere on your system end ink to it

tenten8401, 2018-06-11:
I'd have to re-link it every update as well. In my humble opinion I don't
think a plugin should really be touching the core eclipse install anyways,
but rather a user defined or home folder

corbolais, 2020-08-01:
@jantje 
Thank you for the plugin you wrote and are continuing to improve as a piece of opensource software.

> How would you feel if someone who is using years of work from you for free

I am proud of it. They like it, they use it. Period. This is what FLOSS is all about. That's why I publish my stuff as opensource after all. To contribute, not only by writing my own opensource software but by improving others as well.

Maybe you should reconsider your software licensing. Maybe subscriptionware is more like what you want.

You're linking to a solution to an issue caused by your opensource software behind a paywall. That's what people are hurting themselves on.

cheers

SeongyongPark, 2020-08-04:
Linux Mint 20 is also affected by this, because Eclipse is installed via Flatpak by default
I'm not sure whether you'll have to re-link every time. Still, the use of Flatpak makes it much harder to apply the solution.

@mjwx4p probably shouldn't have gone too far to call the dev "super bad". However, imho, @jantje should also reconsider what F/OSS and github is about.

Issue tracker is there to discuss about the better way of the program, open for everybody. If you can't afford opinions, openness and low income, maybe you should either find ways to capitalize better, or not be dedicated so much.