Sloeber/arduino-eclipse-plugin #493: ARDUINO_BOARD symbol build error

jjssoftware, 2016-07-01:
Hi hopefully I've found the right place to raise this :) 

I originally created an issue over on the esp8266 arduino project about a build problem related to one of the ep8266 libraries. That issue is: https://github.com/esp8266/Arduino/issues/2211

My Eclipse installation details are:
Arduino Eclipse Extensions 3.0.0.201603312133
Eclipse Platform 4.6.0.v20160606-1342 (Neon - recent upgrade on 22/06)

I suppose this could be a Neon / plugin compatibility problem - do you think it's worth trying to install the nightly which looks like it now has support for Neon?

Thanks
Joe

jantje, 2016-07-01:
​As an independent open source project we value your contribution, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back! 
As a supporter you can drag our attention and prioritize your submission... 
please become a supporter!​
https://www.patreon.com/jantje

jjssoftware, 2016-07-02:
Thanks, after a fairly lengthy investigation I've come to the conclusion that we're all good here and the ArduinoGnuMakefileGenerator code is doing the right thing in this plug-in. Keep up the good work!

jantje, 2016-07-02:
Cool you contributed by confirming this is not an issue :-) Thanks. Good work.
As you reopened esp8266/Arduino#2211 I assume you will document the conclusions of your lengthy investigation there.

jjssoftware, 2016-07-02:
Yes I've updated https://github.com/esp8266/Arduino/issues/2211 with the details. 

IMO the problem I saw is just caused by missing escaped quotes in the recipe config. The plug-in simply passes this along through the generated make file and onto the gcc pre-processor which then generates invalid code in the object file resulting in the build failure.

If the ESP8266 arduino core guys agree with this analysis, I'll create the pull request over there.

Thanks
Joe

jantje, 2016-07-02:
I'm closing this one and I'll follow up in esp8266/Arduino#2211

jantje, 2016-07-02:
I'm reopening as this seems to be related to the different command line environment between the ArduinoIDE and the arduino eclipse plugin called sloeber

jantje, 2016-07-02:
for your information this is very simular to #263

jjssoftware, 2016-07-02:
Thanks I'll take a fork and see what I can do about fixing this. If it's similar to https://github.com/jantje/arduino-eclipse-plugin/issues/263 then hopefully it's just a config update rather than a code change..which suits me because I'm no java developer :)

jantje, 2016-07-02:
As you see in #263 it is a windows issue. As you are on Linux I'm a bit confused (see ESP issue for more detail).
Anyway if ESP implements this the Arduino IDE way or their own way I guess something will be needed to make it work.
For #263 I opted to add a "custom config line" (Note that I changed the config change to a lots simpler thing later) but for this issue it first needs to be clear on which oses and which ide's the problem exists.

jjssoftware, 2016-07-02:
I've responded to your last comment in https://github.com/esp8266/Arduino/issues/2211

I'm afraid I can't test anything on windows here but my testing suggests this does seem fine in the Arduino IDE (I've only tested Arduino IDE v1.6.8 on linux) and the problem is localized to this plugin.

Incidentally I've got a fix that seems to be working fine here and it is a config change in this project but we do need to somehow test it on windows

jantje, 2016-07-02:
As workaround you have 2 options:
1) use version 2.2 and not 2.3.
2) add the following line to the project properties->arduino->compile options->append to C and C++
-DARDUINO_BOARD=\"[your string without spaces]\"
If you want spaces: on windows add
-DARDUINO_BOARD="\"[your string with spaces]\""
on linux add (not tested)
'-DARDUINO_BOARD=\"[your string with spaces]\"'

jantje, 2016-07-02:
As a fix following regular expressions are needed
For all oses one that replaces
-D[any char but no whitespace]="[anychar but no whitespace]"
with 
-D[any char but no whitespace]=\"[anychar but no whitespace]\"

for windows one that replaces
'-D[any char but no whitespace]="[anychar]"'
with
-D[any char but no whitespace]="\"[anychar]\""

As regular expressions are not my thing help would be appreciated.

jjssoftware, 2016-07-03:
I was using this change to the post_processing_platform_default.txt file as the fix with a locally rebuilt plug-in:
`build.board.linux=-DARDUINO_BOARD=\"{build.board}\"`
but that's not the fix you want - I agree it is a limited scope fix.

Your workaround works well for the moment until we workout the regex. I'll take a look at that - I assume you'll be wanting java flavoured regex?

jjssoftware, 2016-07-03:
Here's the regex expression that matches:
-D[any char but no whitespace]="[**anychar but no whitespace**]"
`-D\S+="\S+"` (I think the equivalent java string for this is: `"-D\\S+=\"\\S+\""`)

Here's the regex expression that matches:
-D[any char but no whitespace]="[**anychar**]"
`-D\S+=".+?"` (I think the equivalent java string for this is: `"-D\\S+=\".+?\""`)

Tested here:
1. http://www.regexr.com/ 
2. http://www.regexplanet.com/advanced/java/index.html

With this sample text:
`"/home/joe/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -E -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/tmp/build03c94b39c417cc2a52f170a48e8b6a78.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src" "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS" "/tmp/build03c94b39c417cc2a52f170a48e8b6a78.tmp/sketch/.ino.cpp" -o "/tmp/build03c94b39c417cc2a52f170a48e8b6a78.tmp/sketch/.ino.cpp.o"`

jantje, 2016-07-03:
>I was using this change to the post_processing_platform_default.txt file as the fix with a locally rebuilt plug-in:
build.board.linux=-DARDUINO_BOARD=\"{build.board}\"
but that's not the fix you want - I agree it is a limited scope fix.

It is the same thing. Just another place. There are lots of users for whom it is better to stick to the gui.

I tried the regexes you proposed but unfortunately I have some problems.
In the java page the \S is not recognized as everything but white space.
Secondly -and most importantly- the replace part is missing.

jjssoftware, 2016-07-03:
How about this then:
https://gist.github.com/jjssoftware/380664afec598a9056cbd29e2bce4217

Compile / execute / see it working here:
http://goo.gl/uAO3ao

I'm no java developer and this might suit your code style but it does seem to work.

jantje, 2016-07-03:
I've found it :-) 
regex ( -D\S+=)"(\S+)" replace $1\\"$2\\"
With the () you can define "captured groups" and then use these as $x in the replace string

jjssoftware, 2016-07-03:
Well done, not a clue if that works in java. Good luck

jantje, 2016-07-03:
Why did I close this item?
Anyway I think there are following scenario's in the platform and boards.txt file
'-D[defineName]="[define Value]"'
-D[defineName]="[define Value]"
-D[defineName]=\"[define Value]\"
[defineName] can not contain spaces; [define Value] may -but does not have to- contain spaces.

In linux this will always work on the command line
'-D[defineName]="[define Value]"'

In Windows this will always work on the command line
-D[defineName]=\"[define Value]\"

With all provided info I can handle that

jjssoftware, 2016-07-03:
I agree with you analysis on the variants of the types of -D defines that source from boards.txt / platform.txt

> In linux this will always work on the command line
'-D[defineName]="[define Value]"'

I don't agree with this statement at all. This is exactly how -DARDUINO_BOARD is currently defined and it caues a build failure with this plugin on linux. This is the problem I initially reported. 

For me on linux (mint 17.3) if a #define appears on the command line like this, eclipse arduino plugin build attempt fails:
`-DARDUINO_BOARD="ESP8266_ESP01"`

If a #define appears on the command line like this, eclipse arduino plugin build succeeds:
`-DARDUINO_BOARD=\"ESP8266_ESP01\"`

I don't know how to describe it any better than this. Is it clear?

jjssoftware, 2016-07-03:
If it helps at all, I'm happy to test a localized patch to the stable branch code. I'll test on linux and I'll let you know whether it works or not.

jantje, 2016-07-03:
Did you notice the **single quotes around the whole statement**?
have you tried adding this to the C/C++ command line extension?
` '-DARDUINO_BOARD="ESP8266_ESP01"'`
and this
` '-DARDUINO_BOARD="ESP8266 ESP01"'`
As far as I understand they both should work in linux and on mac.

jantje, 2016-07-03:
just tested in windows
`"-DARDUINO_BOARD=\"ESP8266_ESP13\""`
and
`"-DARDUINO_BOARD=\"ESP8266 ESP13\""`
work
Because Linux has the concept of 2 types of string quotes you do not need to backslach the internal quotes

jjssoftware, 2016-07-03:
Sorry my mistake you're right, I missed the outer single quotes. It does work when they're present and the inner double quotes aren't escaped. It also works when the outer single quotes aren't present and the inner double quotes are escaped.

jantje, 2016-07-03:
great. 
the regex has become
` '?-D(\S+)=\\?"(.+?)\\?"'?`
In java that gives 
`" '?-D(\\S+)=\\\\?\"(.+?)\\\\?\"'?"`
As I could test pretty good with the web page you provided I'm pretty confident in the solution I'm building.
Tomorrow the nightly should fix this

jantje, 2016-07-03:
Another problem popped up :-(
Due to the indexer I needed to go for
`"-D[envKey]=\"[env value]\""`
in Windows. I guess the same problem exists in Linux. So; can you check whether
`'-D[envKey]=\"[env value]\"'`
Works fine in Linux?
Try to see what appears when you hover over ARDUINO_BOARD if it is 100% fine it should be "ESP8266 ESP13" but I expect it to be "ESP8266

jjssoftware, 2016-07-03:
Testing via eclipse->project properties->arduino->compile options->append to c and c++..

When I input:

- `'-DARDUINO_BOARD="ESP8266 ESP01"'` == successful build
- `-DARDUINO_BOARD=\"ESP8266 ESP01\"` == successful build
- `'-DARDUINO_BOARD=\"ESP8266 ESP01\"'` == broken build

The final case results in broken pre-processor generated code in the target compiled .o object file that looks like this:

`addServiceTxt("arduino", "tcp", "board", \"ESP8266 ESP01\");` - the escaping backslashes have been incorrectly carried through.

The hover trick you've suggested won't work in any IDE because this symbol is being defined at build time. If the symbol is #define(d) in code, the hover trick would work as you've described.

jjssoftware, 2016-07-03:
Now I've done this I'm convinced this is the best way to test.
I'll go again modifying platform.txt for the same effect

jjssoftware, 2016-07-03:
This round of testing with the following board name defined in boards.txt
`generic.build.board=ESP8266 ESP01`

and 3 separate build attempts with the following setup in platform.txt:

- `'-DARDUINO_BOARD=\"{build.board}\"'` == broken build, escaping backslashes have been incorrectly carried through into compiled code
- `-DARDUINO_BOARD=\"{build.board}\"` == broken build fails early with: error: ESP01": No such file or directory - no code is compiled at all
- `'-DARDUINO_BOARD="{build.board}"'` == build does complete successfully and this is the case where the symbol looks truncated at the space as you described: "ESP8266

Spaces within a defined string literal really are very problematic.

jantje, 2016-07-03:
I made a mistake. This format is the good one for Linux
'-DARDUINO_BOARD="{build.board}"'
It turns out to be lots more complicated then I thought. I'm not sure I'll make it for the next build.

jantje, 2016-07-04:
I just checked on my system. It seems to be fixed on windows.
Can you check with the nightly on linux?

jjssoftware, 2016-07-04:
It looks close but unfortunately not quite right. Here's the command line for the compile of one file from the console output along with the warning that follows:

`"/home/joe/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/home/joe/eclipseWorkspace/mDnsTest/Release/core" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC  -DARDUINO=10609 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266'-DARDUINO_BOARD="ESP8266_ESP01"' -DESP8266  -I"/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" -I"/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" -I"/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS" -I"/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi" -I"/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o"  -Wall
<command-line>:0:21: warning: ISO C99 requires whitespace after the macro name [enabled by default]`

If we look closely, it looks like the warning is coming from no space being present between:
`-DARDUINO_ARCH_ESP8266'-DARDUINO_BOARD="ESP8266_ESP01"'`

The end result is a successful build but no #define for the ARDUINO_BOARD symbol.

jantje, 2016-07-04:
I forgot the space :-( I guess that the compiler ignores the define after the missing space.

jjssoftware, 2016-07-04:
No worries I'll go again when the next build is ready

jjssoftware, 2016-07-05:
**Setup**
- Eclipse IDE for C/C++ Developers 4.6.0.20160613-1800
- Arduino Eclipse Extensions 3.0.0.201607050208 (nightly)
- Linux Mint 17.3

**Test case 1**
boards.txt configuration: generic.build.board=ESP8266_ESP01
results:
1. successful build
2. ARDUINO_BOARD symbol correctly configured

**Test case 2**
boards.txt configuration: generic.build.board=ESP8266 ESP01
results:
1. successful build
2. ARDUINO_BOARD symbol is configured
3. The contents of the compiled ESP8266mDNS.cpp.o object file indicate that the ARDUINO_BOARD symbol is correctly configured as "ESP8266 ESP01"
4. Hovering over the ARDUINO_BOARD symbol in eclipse appears to indicate the symbol is truncated on macro expansion as follows: "ESP8266

For test case 2 item (4), further actual real use of the ARDUINO_BOARD symbol is required to determine if the symbol is or is not truncated.

jantje, 2016-07-05:
>For test case 2 item (4), further actual real use of the ARDUINO_BOARD symbol is required to determine if the symbol is or is not truncated.

I think it is a presentation issue only. I say so because nor codan nor the compiler complain about the code below when compiling the due (where USB_PRODUCT="Arduino due"):
```
#include "Arduino.h"
//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
	Serial.println(USB_PRODUCT);
//Add your repeated code here
}
```

jjssoftware, 2016-07-05:
I'm sure you're probably right but at the same time I'm sure you wouldn't want me to sign off on this on the basis of an assumption because that's not how software testing works.

I see your code and it tells me nothing.

I have similar code that also compiles successfully and that also doesn't 100% confirm that the symbol is fully populated and not truncated in some way.

I'll actually test it in real world actual use and report back soon.

jjssoftware, 2016-07-05:
Confirmed as fixed when tested on linux mint 17.3 in nightly 3.0.0.201607050208
Thanks

rlogiacco, 2016-07-11:
@jantje can we close this?

jantje, 2016-07-12:
@rlogiacco As it is not yet in the stable release I prefer keeping this one open so people can see what is "wrong" in the stable release.
Maybe we should make a link in Sloeber that shows what has been fixed since the running release and what has been reported and confirmed as an issue. I'll think about this one.

overtone1000, 2016-07-26:
Encountered this error when upgrading to 2.3 for the new ESP8266HTTPUpdateServer library. Thanks for sorting this out! Updating to the new nightly build quickly resolved this problem.

jantje, 2016-07-26:
thanks for looking for an existing issue and confirming the solution :-)

closed 2016-08-01 by jantje

danielfmo, 2018-04-11:
I'm getting this issue again in Sloeber 4.2.0.201804050109 and ESP 2.4.0
Does anyone else confirm the issue?
It works in Arduino IDE 1.8.5

C:\Users\...\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS\ESP8266mDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
<command-line>:0:15: error: 'ESP8266_NODEMCU' was not declared in this scope
C:\Users\...\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS\ESP8266mDNS.cpp:902:44: note: in expansion of macro 'ARDUINO_BOARD'
   addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
                                            ^
libraries\ESP8266mDNS\subdir.mk:18: recipe for target 'libraries\ESP8266mDNS\ESP8266mDNS.cpp.o' failed
make: *** [libraries\ESP8266mDNS\ESP8266mDNS.cpp.o] Error 1

jantje, 2018-04-11:
Please do not open old closed issues.
There is indeed a change in the nightly in regards to this.
Open a new issue and I'll tell you how to fix it.