Sloeber/arduino-eclipse-plugin #494: V3 Doesn't Show up

bigdman1216, 2016-07-01:
I have downloaded the plugin but there are no menus and the preferences page does not show "arduino" as an option.  I had a v2 version but cannot revert back.  Is there a V3 install guide that I can follow, I didn't see one?  
Also I can move to any arduino version necessary. 
Thanks!
[image]

jantje, 2016-07-01:
Have you activated the arduino perspective?

bigdman1216, 2016-07-01:
Yes, its empty.  Before when I had V2 I could get through everything except the upload would fail.  I figured it was something to do with my IDE version as per[http://eclipse.baeyens.it/installAdvice.shtml](url)
[image]

jantje, 2016-07-01:
What eclipse version do you run?

bigdman1216, 2016-07-01:
Mars 4.5.2.  Should I download the full product?  Can I overlay my PyDev work on that build?

jantje, 2016-07-01:
>Mars 4.5.2

This should be fine

> Can I overlay my PyDev work on that build?

yes (if you need mars take the stable product; the nightly is based on neon)

Don't you see errors in the error log? Something is seriously wrong in your install

bigdman1216, 2016-07-01:
I downloaded the stable plugin here:  http://www.baeyens.it/eclipse/V3 but it doesn't show.  I am downloading the product version but I have ROS stuff that I'd prefer not to have to do all over again.  
where would I see this error log?  Usually I'll look at the console output to see the issues/warnings/errors.  What else can I show to try and demonstrate the system settings?
thanks

bigdman1216, 2016-07-02:
The full product works.  I'll go load all my other equipment on.  This plugin looks great.  I'm excited to check it out.

rlogiacco, 2016-07-02:
The Arduino perspective you had selected was the previous _V2 related_ perspective. That perspective is a leftover due to perspective id change occurred in Toddler (aka V3). You should be able to open another Arduino perspective (same name, same icon, but diferent id) and delete the old Arduino perspective. The indication is a dangling perspective comes from the double angular quotes surrounding the name and the missing icon.

You can also remove the orphaned perspective to clean up your workspace, if you wish so.

jantje, 2016-07-05:
@bigdman1216 
Is this fixed now?

bigdman1216, 2016-07-05:
Yes, I downloaded the product and it works.  I'll set up patron support
tomorrow.  :) great work.

jantje, 2016-07-05:
Thanks
I'll close this issue then.

closed 2016-07-05 by jantje