Sloeber/arduino-eclipse-plugin #495: No Serial port listed

mandelllee, 2016-07-07:
I'm using v3 (the product version) on a mac running El Capitan. With a node MCU board plugged in and selected there are no serial ports in the drop down list to select. With the exact same configuration with the arduino IDE the serial port is listed and is /dev/cu.SLAB_USBtoUART.

jantje, 2016-07-07:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

mandelllee, 2016-07-07:
I just became a supporter.

Lee

jantje, 2016-07-07:
cool thank you :-)
Normally a problem like this has to do with the filter used in the plugin and the name of the device not matching
In the stable release V3.0 this is
`portNames = SerialPortList.getPortNames("/dev/", Pattern.compile("^cu\\..*(serial|usb).*")); `
As your USB is capital USB this will not match
In the nightly it is:
`portNames = SerialPortList.getPortNames("/dev/", Pattern.compile("^cu\\..*(UART|serial|usb).*")); `
Which should match your device name due to the added UART.
This has changed as part of #455 
As we are close to releasing V3.1 (expected 1th of august) I suggest you upgrade to the nightly.

mandelllee, 2016-07-07:
Does a capital USB have to be added to the pattern or is it just the capital UART that is necessary?

Thank you for working on this so quickly. Also, what time do the nightly builds usually get posted?

Lee

jantje, 2016-07-07:
Capital UART has been added a long time ago. SO the current nightly should work for you.
If you can change the name I would go for making the USB a small usb.
The nightly's get posted after the build so it depends a bit. The build starts at 3 am GMT+1

mandelllee, 2016-07-08:
Hi,

I was wondering if this update made it to the nightly builds. I can’t seem to get to the main nightly builds page and the one for the MacOS has 2016/7/5 as the latest nightly build.

Thank you,

Lee

jantje, 2016-07-08:
The nightly only gets updated if there is a actual change in the code.
The fix you need has been fixed the 19th of may. So 5 July build contains it.

rlogiacco, 2016-07-11:
@mandelllee can you confirm the nightly build fixed this issue for you?

mandelllee, 2016-07-11:
Yes, I was able to install the nightly build and was able to see the serial port.

Thank you,

Lee

jantje, 2016-07-12:
@mandelllee thanks for the feedback

closed 2016-08-01 by jantje