Sloeber/arduino-eclipse-plugin #496: IDE fails to startup

wothke, 2016-07-09:
Today I installed the Win 32-bit version that I had just downloaded here: http://eclipse.baeyens.it/stable.php?OS=Windows . Initially the IDE started and after it had completed the automatic download of the add-on libs. I quit the IDE and copied the "MegaCore" folder that I need for my "ATmega128" microcontroller to "arduinoPlugin/packages".
I then started the IDE and created a new Sketch (*.cpp). The IDE seemed to correctly recognize the new core and let me select the respective ATmega128 settings. So everything seemed fine and it looked as if I had a new empty project. 
I then restarted the IDE - only it does no longer start :-( I get to the point were a popup dialog informs me "we have found programs in the path that might conflict with our external builder" - but that message had not prevented the IDE from starting before.. (see above). but right after that dialog the IDE just crashes - see log below:  
`#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x6d20a299, pid=5692, tid=4528
#
# JRE version: Java(TM) SE Runtime Environment (8.0_66-b18) (build 1.8.0_66-b18)
# Java VM: Java HotSpot(TM) Client VM (25.66-b18 mixed mode windows-x86 )
# Problematic frame:
# V  [jvm.dll+0xaa299]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread (0x003e9800):  JavaThread "main" [_thread_in_vm, id=4528, stack(0x00030000,0x00130000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x0000000c

Registers:
EAX=0x00000000, EBX=0x0012cc94, ECX=0x00000000, EDX=0x000003ff
ESP=0x0012be0c, EBP=0x0012be0c, ESI=0x003e9800, EDI=0xffffffff
EIP=0x6d20a299, EFLAGS=0x00010202

Top of Stack: (sp=0x0012be0c)
0x0012be0c:   0012c258 6d1b747d 00000000 000000b6
0x0012be1c:   0012cc94 003e9800 0012be3c 6d1e1081
0x0012be2c:   12c6e9c8 0012bfc8 6d1e10fc 003e9800
0x0012be3c:   0012bf68 6d1e371e 12c70e30 089186c0
0x0012be4c:   039dbfa0 003e9800 12c6e9c8 003e9800
0x0012be5c:   00000000 003e9800 6d1e373c 00000000
0x0012be6c:   003e9800 6d1e3749 12c6e9c8 0012beb4
0x0012be7c:   009e0417 039dc3b8 00000000 003e9800 

Instructions: (pc=0x6d20a299)
0x6d20a279:   cc cc cc cc cc cc cc 55 8b ec 8b 45 08 8b 4c 81
0x6d20a289:   2c 89 4d 08 8b 45 08 8a d0 f6 d2 f6 c2 01 74 07
0x6d20a299:   8b 40 0c 5d c2 04 00 83 e0 fe 5d c2 04 00 cc cc
0x6d20a2a9:   cc cc cc cc cc cc cc 55 8b ec 8b 45 0c 8b 4c 81 

Register to memory mapping:

EAX=0x00000000 is an unknown value
EBX=0x0012cc94 is pointing into the stack for thread: 0x003e9800
ECX=0x00000000 is an unknown value
EDX=0x000003ff is an unknown value
ESP=0x0012be0c is pointing into the stack for thread: 0x003e9800
EBP=0x0012be0c is pointing into the stack for thread: 0x003e9800
ESI=0x003e9800 is a thread
EDI=0xffffffff is an unknown value

Stack: [0x00030000,0x00130000],  sp=0x0012be0c,  free space=1007k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
V  [jvm.dll+0xaa299]
V  [jvm.dll+0x5747d]
V  [jvm.dll+0x59ca8]
V  [jvm.dll+0x5ac39]
V  [jvm.dll+0x5ad74]
V  [jvm.dll+0xb4ded]
V  [jvm.dll+0xbcd75]
V  [jvm.dll+0xbe978]
V  [jvm.dll+0x10aee6]
V  [jvm.dll+0x10b330]
j  java.lang.Class.getDeclaredConstructors0(Z)[Ljava/lang/reflect/Constructor;+0
[... 495 lines cut ...]
java_command: <unknown>
java_class_path (initial): N:\Program Files\ArduinoWindows\eclipseArduino\plugins\org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
Launcher Type: generic

Environment Variables:
PATH=N:/Program Files/Java/jre8/bin/client;N:/Program Files/Java/jre8/bin;N:/Program Files/Java/jre8/lib/i386;n:\Program Files\Git\git-cheetah\..\bin;n:\Program Files\Git\git-cheetah\..\bin;n:\Program Files\Git\git-cheetah\..\bin;n:\Program Files\Git\git-cheetah\..\bin;n:\Program Files\Git\git-cheetah\..\bin;n:\Program Files\Git\git-cheetah\..\bin;C:\Documents and Settings\All Users\Application Data\Oracle\Java\javapath;n:\sniff\bin;C:\Program Files\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\system32\WindowsPowerShell\v1.0;n:\Program Files\Emscripten\clang\3.2_32bit\bin;n:\Program Files\Emscripten\node\0.10.17_32bit;n:\Program Files\Emscripten\python\2.7.5.3_32bit;n:\Program Files\Emscripten\java\7.45_32bit\bin;n:\Program Files\Emscripten;n:\Program Files\Emscripten\emscripten\1.12.0;n:\Program Files\Emscripten\crunch\1.03;n:\Program Files\Emscripten\mingw\4.6.2_32bit;N:\Program Files\Git\bin;D:\Program Files\PHP;c:\Windows\system32;N:\Program Files\ArduinoWindows\eclipseArduino;
USERNAME=wothke
SHELL=n:\sniff\bin\sh.exe
OS=Windows_NT
PROCESSOR_IDENTIFIER=x86 Family 15 Model 4 Stepping 10, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 5.1 Build 2600 (5.1.2600.5512)

CPU:total 2 (1 cores per cpu, 2 threads per core) family 15 model 4 stepping 10, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ht, tsc

Memory: 4k page, physical 3734636k(2383528k free), swap 5138212k(3811428k free)

vm_info: Java HotSpot(TM) Client VM (25.66-b18) for windows-x86 JRE (1.8.0_66-b18), built on Nov  9 2015 10:58:29 by "java_re" with MS VC++ 10.0 (VS2010)

time: Sat Jul 09 19:09:34 2016
elapsed time: 17 seconds (0d 0h 0m 17s)

`

jantje, 2016-07-09:
Well documented issue. :-)
So here are my thoughts:
First off all: 
You should **not** mess in the folder arduinoPlugin or any of its subfolders.
To add other hardware add a location to windows->preferences->arduino->private hardware path.

Secondly:
I advice you to be more aggressive towards your path variable in windows. There is no need to have "N:\Program Files\ArduinoWindows\eclipseArduino" in your path. Nor is there a reason to have multiple times the same path there "n:\Program Files\Git\git-cheetah..\bin;n:\Program Files\Git\git-cheetah..\bin;n:\Program Files\Git\git-cheetah..\bin;n:\Program Files\Git\git-cheetah..\bin;n:\Program Files\Git\git-cheetah..\bin;n:\Program Files\Git\git-cheetah..\bin;"
I say so because of thirdly

Thirdly:
The startup problem you have now is probably a duplicate of issue #415 
Switching to the nightly will remove the crashing but not the problem.
So even when you are running the nightly you will still have to modify your path before starting sloeber. This is so because sh and bash are known to give problems when they are in the path.  
2 entries in your path I suspect of problems N:\Program Files\Git\bin; n:\Program Files\Emscripten\mingw\4.6.2_32bit.
Use where sh and where bash to find the "conflicting folders"

lastly:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Best regards
Jantje

wothke, 2016-07-09:
firstly: thx, I moved the folder accordingly
secondly: when printing my PATH in the CMD window it does NOT contain "N:\Program Files\ArduinoWindows\eclipseArduino" at all, nor does it contain multiple ..git-cheetah.. entries. I'd say we are looking at some IDE bug with regard to how it compiles that path.
thirdly: actually it's SNIFF+ and GIT that have sh.exe and/or bash 

having changed 1 + 3 the IDE no longer crashes on startup nor does it show the popup warning

lastly: lets first wait and see if I actually like the project.. my first impression is that it does not use my configured private library path nor does it find some standard Arduino stuff like unit16_t..

jantje, 2016-07-09:
>my first impression is that it does not find standard Arduino stuff like unit16_t..

It should but for simple sketches (which means you do not have includes) you need to include Arduino.h.
I have an article explaining in depth but it is patron's only https://www.patreon.com/posts/there-are-bugs-5507246
Otherwise right click the project and select indexer rebuild.

>PS: where is the "Upload Using Programmer" functionality?

In the nightly

wothke, 2016-07-10:
My code is already including "Arduino.h". But when I look at that file (core/core/Arduino.h) the problem is actually that the IDE does not find the various includes that are made from that file (e.g. stdlib.h, stdbool.h, string.h, math.h, avr/pgmspace.h, etc). It seems that the Win32 installer did not create a consistent environment where the various standard includes can be located properly. (I did not see any hints that there is stuff that needs to be installed separately to make V3 work - is there?)

jantje, 2016-07-10:
>I did not see any hints that there is stuff that needs to be installed separately to make V3 work - is there?

Toddler works out of the box. You did something wrong somewhere.

closed 2016-07-10 by jantje

wothke, 2016-07-10:
LOL. IDE does not find standard includes right after fresh install.. sure must be my mistake. 

lastly: yes it seems I did something wrong: should never have installed this IDE - but that can be fixed!

wothke, 2016-07-11:
update: Actually there is a problem with the Git-cheetah shell extension: On WinXP (and maybe elsewhere) it automatically keeps adding respective "Git-cheetah/../bin" entries to the PATH. As soon as one enters the "Environment Variables" dialog these added entries are removed but after a while they will start to reappear. (i.e. sh.exe / bash.exe conflicts are automatically re-introduced after a while..)

After completely removing the Git-cheetah extension, the build problems of the empty standard "ProMini" project (see above post) have disapeard. However the build for some add-on hardware (i.e. MegaCore) still fails with the same kind of error message.

jantje, 2016-07-11:
> update: Actually there is a problem with the Git-cheetah shell extension: On WinXP (and maybe elsewhere) it automatically keeps adding respective "Git-cheetah/../bin" entries to the PATH. As soon as [...]

Sounds like something I would never ever want to install on a system.
Anyway seems like this problem is fixed and we know  Git-cheetah shell extensions are incompatible with Sloeber.

rlogiacco, 2016-07-11:
> However the build for some add-on hardware (i.e. MegaCore) still fails with the same kind of error message.

Did you clear the the `arduinoPlugin` folder from your modifications following @jantje suggestion to add a separate hardware folder?
I don't know the _MegaCore_ board you mention, which toolchain does it use?

wothke, 2016-07-11:
I had made the mistake to manually copy the MegaCore (see https://github.com/MCUdude/MegaCore) folder from my existing Arduino IDE installation. After properly reinstalling it using the .JSON the build problems have disappeared. thx.

rlogiacco, 2016-07-11:
Good @wothke , feel free to come back and tell us about your experience with _Sloeber_ once you have tried it properly! Also stay around for the upcoming release and, in case everything works well for you, remember your support is welcome, whatever format you decide to pick!

@jantje don't know if you wish to apply any label to this issue to mark it resolved

jantje, 2016-07-12:
>@jantje don't know if you wish to apply any label to this issue to mark it resolved

@rlogiacco The issue is closed as duplicate. As the original issue is a duplicate. The second issue is usage error. So I guess it is labelled as good as possible.