Sloeber/arduino-eclipse-plugin #497: Upload protocol settings ignored

DemoN3579, 2016-07-11:
I'm usig Eclipse IDE for Arduino Developers with Arduino Eclipse Extensions 3.0.0.201607090208 and try to upload sketch to my Leonardo Eth board through AVRISP STK500.
I've set board : Arduino Leonardo, upload protocol : Atmel STK500 development board and correct usb port for programmer.
Then I try to upload sketch IDE try to use default way for the Leonardo and not use programmer. In upload console I see 

Starting upload
using arduino loader
Starting reset using 1200bps touch process
PORTS { /dev/ttyACM0, /dev/ttyUSB0,} / { /dev/ttyACM0, /dev/ttyUSB0,} => {}
PORTS { /dev/ttyACM0, /dev/ttyUSB0,} / { /dev/ttyACM0, /dev/ttyUSB0,} => {}

If I use Arduino IDE with te same programmer it's work well.

rlogiacco, 2016-07-11:
Which OS are you using?

DemoN3579, 2016-07-12:
Ubuntu 16.04 LTS

jantje, 2016-07-12:
this sounds like a duplicate of #479 to me.
From what is shown the leonardo is not resetting. We are not even at the programmer action yet.

DemoN3579, 2016-07-12:
I think that it's not duplicate of  https://github.com/jantje/arduino-eclipse-plugin/issues/479
In case of #479 issue it's problem with the board reset.

But I've programmer AVRISP STK500 connected to /dev/ttyUSB0 
I set upload protocol : Atmel STK500 development board end excpect that the IDE will upload sketch through programmer. In this case it's not needed to reset board.
But IDE try to reset programmer like  Leonardo board!

DemoN3579, 2016-07-12:
simply IDE use not that way to upload skecth that I expect
I excpect that  skecth will be uploaded through AVRISP programmer but IDE try do  as if it's simply Leonardo way.
And IDE didn't try tointeract with programmer

DemoN3579, 2016-07-12:
The same situation on linux
Sloeber is trying to reset the port  and after that one failed he is issuing the correct command
I was not so patient first time 

The problem only that sloeber try to reset port first

jantje, 2016-07-12:
patience is a virtue.

jantje, 2016-07-13:
The build failed so I relaunched the build. The fix is now available on the nightly.
Can you check?

DemoN3579, 2016-07-13:
Fixed
everything's work correctly
the sketch uploaded as expected through programmer
thank you

closed 2016-08-01 by jantje