Sloeber/arduino-eclipse-plugin #498: non standard core issues

wothke, 2016-07-11:
In order to build for ATmega128 I am using https://github.com/MCUdude/MegaCore. The respective add-on hardware options are correctly presented when creating a respective new project but there then are the below problems:

1) The automatically generated list of "Includes" only contains the folders of the added core but NONE of the standard Arduino stuff (e.g. ".. avr/include", "..avr/4.8.1/include", "..avr/4.8.1/include-fixed"). Consequently the sketch will not find even the most basic stuff (like "uint8_t"). I guess that it is a bug of the project generation for non-standard/built-in cores that respective include folders are omitted. 

2) Even when adding the missing 3 includes manually (see point 1), there still is a problem that the build does not even start. I suspect that it also might be due to some path not being setup properly. 

For comparison here a section from a correctly working standard-core build command:

`"N:\Program Files\ArduinoWindows\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR     -I"N:\Program Files\ArduinoWindows\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\cores\arduino" -I"N:\Program Files\ArduinoWindows\eclipseArduino\arduinoPlugin\packages\arduino\hardware\avr\1.6.11\variants\eightanaloginputs" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
'Finished building: ../.ino.cpp'`

and here what the IDE is trying to do for the non-standard core:

`"/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR     -I"C:\Documents and Settings\wothke\Arduino\hardware\MegaCore\hardware\avr\1.0.2\cores\MegaCore" -I"C:\Documents and Settings\wothke\Arduino\hardware\MegaCore\hardware\avr\1.0.2\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
subdir.mk:24: recipe for target '.ino.cpp.o' failed`

for some reason the IDE is using a fully qualified path to "avr-g++" in the first case but not in the second case. I don't see why the IDE should behave differently here for standard and non-standard cores.

3) Finally there is an annoying flaw with regards to syntax highlighting in the editor: When specifically creating a project for the "ATmega128", then obviously at compile time __AVR_ATmega128__ will be defined. However that information should be equally known to the IDE's editor (since the target hardware was specifically selected in the project) and respective syntax highlighting should be based on that information. Currently the editor critizices various "issues" that are caused by the fact that the editor does not apply the respective target hardware specific defines.

jantje, 2016-07-11:
Have you installed the board as explained here? https://github.com/MCUdude/MegaCore
more specifically did you use the https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json file with Sloeber?

wothke, 2016-07-11:
mea culpa :-( I had tried a manual install (i.e. copying the folder from the Arduino IDE).. I now reinstalled the MegaCore using the JSON file. Now the above phantom problems all disappeared. sorry to have bothered you..

jantje, 2016-07-11:
Think about supporting us. I think we proved to you we deserve it https://www.patreon.com/jantje

closed 2016-07-11 by jantje