Sloeber/arduino-eclipse-plugin #499: Add information to sloeber to show what has been fixed/reported

jantje, 2016-07-12:
We currently keep issues open till they are released. We do so because the standard query in git is "open issues". As such most people will not find issues that have been fixed later then what they have installed.

Keeping the issues open however has some serious drawbacks. Apart from the fact it looks like we have lots of issues open it is also a problem for the core team as they remain visible in eclipse.

Adding 1 or more links to Sloeber that make following query information available to the user: "issues known in your installation", "issues fixed in the latest stable release" and "issues fixed in the nightly" would allow to take away the requirement of having the issues open in github.

rlogiacco, 2016-07-12:
So, if I understand correctly, you are suggesting to create three web pages in our web site listing _known issues_, _next release features/fixes_ and _current release features/fixes_.

While I do understand the goal, I believe the amount of maintenance and attention required is not going to be trivial: each issue must be marked accordingly to the release into which it is planned to be delivered for this to work with the consequent label count explosion.

May be we can use milestones for this... never tried that feature myself though

jantje, 2016-07-12:
I'm would make one(or more) pages that link to github based on labels. Using php and the labelling in github as we use right now.
The main thing is that the github adds `is:open` by default to the query. By having our own page we can force the correct queries.

jantje, 2016-07-13:
Guess what I got it to work. It looks bad (no surprise)
Try these links http://eclipse.baeyens.it/status.php?majorVersion=3&minorVersion=0&microVersion=0&qualifierVersion=20160102
http://eclipse.baeyens.it/status.php?majorVersion=1&minorVersion=0&microVersion=0&qualifierVersion=20160102
http://eclipse.baeyens.it/status.php?majorVersion=2&minorVersion=3&microVersion=0&qualifierVersion=20160102
It would be better if I could do a or query of labels on github (didn't find how) or if I could include the results on the same page.

rlogiacco, 2016-07-17:
Using `https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues?labels=status: fixed in nightly` along with an authentucation token I generated on my account I managed to obtain the list of issues in JSON format:

```
[
  {
    "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues/497",
    "repository_url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin",
    "labels_url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues/497/labels{/name}",
    "comments_url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues/497/comments",
    "events_url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues/497/events",
    "html_url": "https://github.com/jantje/arduino-eclipse-plugin/issues/497",
    "id": 164778648,
    "number": 497,
    "title": "Upload protocol settings still take reset command into account",
    "user": {
      "login": "DemoN3579",
      "id": 20394106,
      "avatar_url": "https://avatars.githubusercontent.com/u/20394106?v=3",
      "gravatar_id": "",
      "url": "https://api.github.com/users/DemoN3579",
      "html_url": "https://github.com/DemoN3579",
      "followers_url": "https://api.github.com/users/DemoN3579/followers",
      "following_url": "https://api.github.com/users/DemoN3579/following{/other_user}",
      "gists_url": "https://api.github.com/users/DemoN3579/gists{/gist_id}",
      "starred_url": "https://api.github.com/users/DemoN3579/starred{/owner}{/repo}",
      "subscriptions_url": "https://api.github.com/users/DemoN3579/subscriptions",
      "organizations_url": "https://api.github.com/users/DemoN3579/orgs",
      "repos_url": "https://api.github.com/users/DemoN3579/repos",
      "events_url": "https://api.github.com/users/DemoN3579/events{/privacy}",
      "received_events_url": "https://api.github.com/users/DemoN3579/received_events",
      "type": "User",
      "site_admin": false
    },
    "labels": [
      {
        "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/labels/domain:%20upload",
        "name": "domain: upload",
        "color": "bfe5bf"
      },
      {
        "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/labels/importance:%20board%20specific",
        "name": "importance: board specific",
        "color": "0052CC"
      },
      {
        "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/labels/OS:%20all",
        "name": "OS: all",
        "color": "eb6420"
      },
      {
        "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/labels/status:%20fixed%20in%20nightly",
        "name": "status: fixed in nightly",
        "color": "f7c6c7"
      }
    ],
    "state": "open",
    "locked": false,
    "assignee": null,
    "assignees": [

    ],
    "milestone": null,
    "comments": 10,
    "created_at": "2016-07-11T07:49:46Z",
    "updated_at": "2016-07-13T23:42:38Z",
    "closed_at": null,
    "body": "I'm usig Eclipse IDE for Arduino Developers with Arduino Eclipse Extensions 3.0.0.201607090208 and try to upload sketch to my Leonardo Eth board through AVRISP STK500.\r\nI've set board : Arduino Leonardo, upload protocol : Atmel STK500 development board and correct usb port for programmer.\r\nThen I try to upload sketch IDE try to use default way for the Leonardo and not use programmer. In upload console I see \r\n\r\nStarting upload\r\nusing arduino loader\r\nStarting reset using 1200bps touch process\r\nPORTS { /dev/ttyACM0, /dev/ttyUSB0,} / { /dev/ttyACM0, /dev/ttyUSB0,} => {}\r\nPORTS { /dev/ttyACM0, /dev/ttyUSB0,} / { /dev/ttyACM0, /dev/ttyUSB0,} => {}\r\n\r\n\r\nIf I use Arduino IDE with te same programmer it's work well."
  },
  {
    "url": "https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues/495",
[... 264 lines cut ...]
        "gravatar_id": "",
        "url": "https://api.github.com/users/rlogiacco",
        "html_url": "https://github.com/rlogiacco",
        "followers_url": "https://api.github.com/users/rlogiacco/followers",
        "following_url": "https://api.github.com/users/rlogiacco/following{/other_user}",
        "gists_url": "https://api.github.com/users/rlogiacco/gists{/gist_id}",
        "starred_url": "https://api.github.com/users/rlogiacco/starred{/owner}{/repo}",
        "subscriptions_url": "https://api.github.com/users/rlogiacco/subscriptions",
        "organizations_url": "https://api.github.com/users/rlogiacco/orgs",
        "repos_url": "https://api.github.com/users/rlogiacco/repos",
        "events_url": "https://api.github.com/users/rlogiacco/events{/privacy}",
        "received_events_url": "https://api.github.com/users/rlogiacco/received_events",
        "type": "User",
        "site_admin": false
      }
    ],
    "milestone": null,
    "comments": 3,
    "created_at": "2014-05-12T13:53:22Z",
    "updated_at": "2016-05-07T19:56:47Z",
    "closed_at": null,
    "body": "Now the serial monitor always reverts to the default (no linefeed) option.\r\nAdding a \"last used serial monitor option\" would be nice."
  }
]
```

rlogiacco, 2016-07-17:
With the JSON above we can create the list the way we like... But my concern still remains: how do we distinguish the issues `fixed in nightly` for 3.1 from those `fixed in nightly` for 3.2?

Should we use an additional filter by date?
Do you intend to use another label to also mark the version?
Do we use something else?

jantje, 2016-07-17:
@rlogiacco 
I had found the API as well but you are far better at this then I am :-)
This should make the php relatively easy and allow for a nearly completely automated page.

> how do we distinguish the issues fixed in nightly for 3.1 from those fixed in nightly for 3.2?

There is only fixed in nightly. :-)
As described in the wiki https://github.com/jantje/arduino-eclipse-plugin/wiki/The-release-process

>  Label activity in github.
> 
> Rename the label "status: fixed in nightly" to "status:fixed in release X.Y" and create a new label "status: fixed in nightly"

jantje, 2016-09-18:
It doesn't look good but I'm getting there http://baeyens.it/eclipse/github%20status.php

This is my current code
```php
<?php
//echo "start<br>";
$issuesJson = file_get_contents('https://api.github.com/repos/jantje/arduino-eclipse-plugin/issues?labels=status:%20fixed%20in%20nightly');
$issues=json_decode($issuesJson,true);
//var_dump($issues);
foreach ($issues as $issue){
    echo "<a href=" . $issue["html_url"] . ">" . $issue["title"] . "</a> ";
    foreach ($issue["labels"] as $label){
        echo'<SPAN style="background-color:' .$label["color"] . '"> ' . $label["name"] ." </SPAN>&ensp;";
    }
    echo "<br />\n";
}
//echo "end<br>";

?>
```

jantje, 2017-04-01:
The web page still needs to be filtered

closed 2017-05-25 by jantje