Sloeber/arduino-eclipse-plugin #500: upload via ISP programmer

wothke, 2016-07-13:
I am using an ISP programmer to upload sketches to my ATmega128 and with the old ArduinoIDE this works (i.e. my programmer and the respective USB driver works). According to some previous feedback the nighly build supposedly contains "upload using programmer" funtionality. 

I had downloaded the (then) latest nightly build (2016-07-09) but I don't see a respective feature nor does the standard "upload" functionality recognize the programmer. 

Am I missing something?

jantje, 2016-07-13:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

wothke, 2016-07-14:
Just me 2cts: You should give your users a chance to at least once successfully compile + upload a sketch.. (or as @rlogiacco amiably suggested to even "tell us about your experience with Sloeber once you have tried it properly!" ). To constantly nag your prospective users about sponsorship before they've had a change to do so is highly counterproductive.
Unfortunately your earlier claim that the Nightly build would provide the "upload using programmer" functionality just seems to be wrong and therefore it seems that Sloeber is actually useless to me and my ISP programmer :-( too bad, since I woud have loved to dump that crappy ArduinoIDE.

jantje, 2016-07-14:
This is your third issue. I helped you out with your first 2 issues that both were documented.
And programmers is available in the nightly. Proof of that #497
Just me 2cts: You should learn to rtfm or pay for services.

wothke, 2016-07-14:
You are wrong: The "Upload Protocol" selection is available for the built-in boards but for add-on boards (e.g. MegaCore) the respective options list is empty, i.e. it only contains a "Default" entry - NOT one of the different programmers is provided for the add-on board. If you have documented this "feature" then you have indeed hidden that fucking manual very well. 

(I manually hacked the needed "USBasp" programmer into my ATmega128 project configuration and with that programmer the uploading actually works, i.e. I see no reason why the respective programmer should not be presented in the "Upload Protocol" selection. Unfortunately my hack apparently introduced some inconsistency so that the regular "Build project" now throws a NullPointerException - so I won't use that feature if it isn't properly supported by the IDE.)

closed 2016-07-14 by jantje