Sloeber/arduino-eclipse-plugin #502: serial monitor: auto-clear on start/reset

jonbarril, 2016-07-14:
Appending new output to old output is often confusing.  An option to auto-clear the output would be helpful, such as upon upload and/or upon reset.

jantje, 2016-07-14:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jrobert-gh, 2016-08-18:
jantje, I agree this would be a useful addition.

jantje, 2016-08-19:
@jrobert-gh Are you a sponsor@patreon?

jrobert-gh, 2016-08-19:
Yes, as 'JRobert'.

JRobert

jantje, 2016-08-19:
If you support a issue as patron best what you can do is post the issue number here https://www.patreon.com/posts/i-support-issue-3700469.

jantje, 2016-08-19:
I'll have a look at this when I'm in this area. I'm a bit bussy with adding testing right now.

jantje, 2016-08-21:
It turned out to be relatively easy. Can you test after 06:00 GMT?

jrobert-gh, 2016-08-21:
I sure will - thanks!

jrobert-gh, 2016-08-22:
Worked for me. Thanks, Jantje.

jantje, 2016-08-22:
txs for confirming

closed 2017-01-17 by jantje