Sloeber/arduino-eclipse-plugin #503: recipe for target 'arduino.ar' failed

koverg70, 2016-07-15:
After upgrading to latest version:

Starting archiver
"C:\java\eclipseArduino\arduinoPlugin\tools\esp8266\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "C:/java/eclipseArduino/workspace/P1/Release/arduino.ar"  arduino.ar  .\core\umm_malloc.c.o  .\core\spiffs_cache.c.o .\core\spiffs_check.c.o .\core\spiffs_gc.c.o .\core\spiffs_hydrogen.c.o .\core\spiffs_nucleus.c.o  .\core\cdecode.c.o .\core\cencode.c.o  .\core\Esp.cpp.o .\core\FS.cpp.o .\core\HardwareSerial.cpp.o .\core\IPAddress.cpp.o .\core\MD5Builder.cpp.o .\core\Print.cpp.o .\core\Schedule.cpp.o .\core\Stream.cpp.o .\core\StreamString.cpp.o .\core\Tone.cpp.o .\core\Updater.cpp.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\base64.cpp.o .\core\cbuf.cpp.o .\core\cont.S.o .\core\cont_util.c.o .\core\core_esp8266_eboot_command.c.o .\core\core_esp8266_flash_utils.c.o .\core\core_esp8266_i2s.c.o .\core\core_esp8266_main.cpp.o .\core\core_esp8266_noniso.c.o .\core\core_esp8266_phy.c.o .\core\core_esp8266_postmortem.c.o .\core\core_esp8266_si2c.c.o .\core\core_esp8266_timer.c.o .\core\core_esp8266_wiring.c.o .\core\core_esp8266_wiring_analog.c.o .\core\core_esp8266_wiring_digital.c.o .\core\core_esp8266_wiring_pulse.c.o .\core\core_esp8266_wiring_pwm.c.o .\core\core_esp8266_wiring_shift.c.o .\core\debug.cpp.o .\core\heap.c.o .\core\libc_replacements.c.o .\core\pgmspace.cpp.o .\core\spiffs_api.cpp.o .\core\spiffs_hal.cpp.o .\core\time.c.o .\core\uart.c.o 
C:\java\eclipseArduino\arduinoPlugin\tools\esp8266\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-ar.exe: **_.coreumm_malloc.c.o: No such file or directory_**
makefile:76: recipe for target 'arduino.ar' failed
make: *** [arduino.ar] Error 1

jantje, 2016-07-15:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

koverg70, 2016-07-16:
I did not ask for support, onlyreported a bug.

It took more than an hour to reinstall the original version, create a simple empty project, test the compilation. Then I updated to latest version and the compilation of the same project broke. That was my  contribution to the project.

Surely I will support with money when I find it useful for my needs, but until now it just cause me troubles and vasted time. Sorry, I hope this will change in the near future.

Best regards!

jantje, 2016-07-16:
I did not say you asked for support, we ask for support.
Have you followed the update instructions? http://eclipse.baeyens.it/how_to.shtml#/n

rlogiacco, 2016-07-16:
As @jantje said, we welcome your bug and, if confirmed, we'll find our best to fix it. If you wish to see the bug be prioritized than you can decide to support us.
I'm sorry you feel you wasted more than an hour trying to set up the plugin, but please consider the greater amount of time we have spent to create this plugin: if you feel reporting a potential bug is all the contribution you can provide I believe you don't get the point about _community_ and _open source_.

Anyway, the upgrade guide provided by @jantje is proven to work: can you please double check you followed it?

koverg70, 2016-07-17:
Hi Guys!

Thanks for your responses. Actually I found out that the problem was PortableGit on my PATH.
I removed its bin folder from the PATH and everything is OK now.

Although it is true that previous versions of the arduino-eclipse-plugin was not confised by PortableGit on the PATH.

Best regards,

Gábor

closed 2016-07-17 by koverg70

jantje, 2016-07-17:
Thanks for the feedback.
Not sure what you mean with "previous version". But V2 and V3 are majorly different in setup. 

I assume PortableGit contains sh or bash. We only started noticing the bash and sh issue in V3. It does not even have to be plugin related, it could be something in CDT or eclipse that changed as we keep up to date with these once as well.