Sloeber/arduino-eclipse-plugin #505: Board definition parse error

ceebas1981, 2016-07-19:
A json board definition file from digistump gets parsed correctly in the latest Arduino IDE, but fails in the Eclipse plugin. See screenshot below.

[image]

rlogiacco, 2016-07-19:
May be the Arduino IDE it's more lenient on JSON format, but looking at the stack trace it appears the JSON is invalid

rlogiacco, 2016-07-19:
I've checked the file at digistump and it looks correct, can you please post the content of your local copy of the file? I wonder if there has been any download issue...
Please use the `code` formatting button (<>) when inserting the file content: it will make it easier to read.

jantje, 2016-07-19:
This looks like a duplicate of #393 to me.

ceebas1981, 2016-07-19:
Indeed a double to #393. Sorry about that. Thanks.

jantje, 2016-07-19:
Thanks for the feedback

closed 2016-07-22 by jantje