Sloeber/arduino-eclipse-plugin #506: Fresh nightly install on Eclipse Neon, Windows 10

tomkcook, 2016-07-20:
Attempting to install today's nightly on a clean Eclipse Neon CDT installation on Windows 10 leads to a broken toolchain.  After installing the plugin and restarting the workbench, the download of the toolchain hangs and eventually eclipse has to be terminated.  On the next start, the board packages and libraries are downloaded but any build fails because make is not present.

Once v3 has been installed, the nightly can then be installed successfully.  This works even if the plugin is uninstalled and the arduinoPlugin folder deleted; the nightly can then be installed directly without first installing v3.

jantje, 2016-07-21:
I'm sorry but your story is not clear to me.
It looks like there was some internet connection issue which makes the install fail. 
Once that happened your eclipse install is corrupted. Deleting the arduinoPlugin folder (as you did) should fix that issue.
So it seems to me as if sloeber is working fine

tomkcook, 2016-07-22:
I'll try again and try to quantify the problem.

jantje, 2016-07-22:
txs :-)

jantje, 2016-08-20:
Any progress?

jantje, 2016-09-03:
Closing as user error due to lack of input

closed 2016-09-03 by jantje