Sloeber/arduino-eclipse-plugin #507: Feature request: pre enabled color coding

MCUdude, 2016-07-22:
Hi!
I guess it would be really nice for new Slober IDE users not having to figure out the color coding them self, and just enable the "Arduino style" by default. I did some tweaking, and I'm really happy with the result! Is this doable? I belive most users would like the colors to pre-set, and change them if they want to 😉 

Attached is my current GUI configuration file for the Arduino text color coding
[GUI_settings.zip](https://github.com/jantje/arduino-eclipse-plugin/files/379187/GUI_settings.zip)

jantje, 2016-07-22:
I'm pretty sure some people do not want the plugin to change the settings they currently have.
But it sure is something we can "offer"
I myself work nearly always with the defaults, so I'm not a specialist. What would need to be done to get this to work?

MCUdude, 2016-07-24:
Maybe the file can be an optional download? If so, it's important that the user gets informed about the file, and that it's recommended for new users

jantje, 2016-07-24:
I mean I have extracted the file from the zip and then I do what?

rlogiacco, 2016-07-25:
@jantje you have to right click on the project pane and select _import_ then select _preferences_ and then pick the `.epf` file you extracted. be aware though, the settings will apply to all CDT, meaning we are heare talking about setting the CDT preferences for C/CPP syntax highligthing.

I personally prefer much more the Eclipse defaults, so my personal vote for these defaults is :-1:

jantje, 2016-07-25:
@rlogiacco 
Thanks: I'll give it a try.
I agree that the advanced users will not want this as default. However for starters it may add value so bundling it and asking at the very first start -of the product- may be a good solution.
What do you mean with "I believe that file might end up into a contrib section, eventually...."

MCUdude, 2016-07-25:
How can the file be made user independent? I'm able to find my computer user name and main work directory multiple places in that file. It's not a binary file, so it can be edited with any text editor 😉

jantje, 2016-07-25:
@MCUdude 
I tough you were the expert :-s

rlogiacco, 2016-07-26:
> I believe that file might end up into a contrib section, eventually....

I'm not a fan of _questions at the beginning_ approach myself, especially for newcomers who might don't even clearly understand what the question is about and what would be the outcome of their answer, or if it can be revertable...

So I believe we might consider having a section of the website where such contributions can be downloaded and applied, with each contribution explaining what it does and how to apply it... But this makes sense only if there are going to be more than a couple of them.

If the all thing boils down to one option, then it's probably easier to just have a switchable setting, but whatever it is going to be, be aware it will apply to CDT as a whole, not just to Arduino (this note is *not* for @jantje who is well aware of this).

jantje, 2016-07-26:
Tried it. It worked. 
I think best is to describe in the how to's how you can use this file to get the "Arduino IDE look"

jantje, 2016-08-20:
I've had some more thinking about this. Could we export these setting with a default installation? If so we can put the "default" and "arduino" style on the web site. This way users could revert back.

closed 2019-01-09 by jantje