Sloeber/arduino-eclipse-plugin #508: duplicate icons

cho934, 2016-07-27:
Hi, I'm trying now to use v3, but idem to v2, sometimes when I open Eclipse (Luna or Mars), all arduino icons are duplicated on all perpective again and again. Very strange, and very annoying.

If I customize the perspective, I can only see each icon but without duplication, If I remove all of them, all icons disappear, but if I add some of them, each icon keep duplicated.

What's wrong ?

rlogiacco, 2016-07-27:
Can you please specify which version exactly are you installing? 
I believe you are speaking of the plugin and not of the bundled product, but is it the stable version or the nightly? And which version of the nightly, eventually?

cho934, 2016-07-27:
Eclipse Luna and I had Arduino Eclipse plugin version 2. 
I'm working on a virtual machine with CUbuntu 14.04.
I've just migrate Eclipse to Mars.2 and installed Arduino Eclipse plugin version 3. 
But I've just noticed that the plugin version 2 is always there also. So the Bug from version 2 is always there... 
I will uninstall all versions and install only v3.

cho934, 2016-07-27:
If I uninstall all then reinstall V3 plugin from Eclipse Market place, V2 is also installed and the problem of duplicate icons is always there...

[image]

jantje, 2016-07-27:
If you have V2 and V3 installed it is normal that you have the icons double.

cho934, 2016-07-29:
As soon as I installed V3, V2 is also installed using the market place. If I understand your remark, it is in fact a problem...

jantje, 2016-07-29:
V2 and V3 have a different update site and V3 is marked as an update of V2 and you are the first one to report this problem. So there must be some "difference" between "your system" and "my system"
Have you activated oomph
Do you have the same problem with the product?

jantje, 2016-08-20:
@cho934 Have you been able to find the root cause?

jantje, 2016-09-03:
Closing as user error due to lack of input.

closed 2016-09-03 by jantje