Sloeber/arduino-eclipse-plugin #509: pins_arduino.h not found when more than one pinout file is available

MCUdude, 2016-07-27:
Hi!
I'm testing my Arduino cores with this amazing plugin, but I've found a potential issue with my [MightyCore](https://github.com/MCUdude/MightyCore), since this core actually have two pins_arduino.h file. There is one generic file and one with the "Bobuino", which is a UNO approach.
When using this core I get this error when trying to compile a blank sketch.

``` bash
17:31:42 **** Incremental Build of configuration Release for project tooldeps ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/tools/arduino/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   -I"/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o"  -Wall
In file included from ../.ino.cpp:7:0:
/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore/Arduino.h:255:26: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
                          ^
compilation terminated.
make: *** [.ino.cpp.o] Error 1

17:31:42 Build Finished (took 273ms)
```

Have you seen this before?

MCUdude, 2016-07-27:
blink.ino, Arduino 1.6.10, OSX 10.11.5, ATmega1284P

``` bash
/Users/Hans/Downloads/Arduino-1.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/hardware" -hardware "/Users/Hans/Library/Arduino15/packages" -hardware "/Users/Hans/Documents/Arduino/hardware" -tools "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/tools-builder" -tools "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/hardware/tools/avr" -tools "/Users/Hans/Library/Arduino15/packages" -built-in-libraries "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/libraries" -libraries "/Users/Hans/Documents/Arduino/libraries" -fqbn=MightyCore:avr:1284:pinout=standard,variant=modelP,BOD=2v7,clock=16MHz_external -ide-version=10610 -build-path "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/Hans/Downloads/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
/Users/Hans/Downloads/Arduino-1.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/hardware" -hardware "/Users/Hans/Library/Arduino15/packages" -hardware "/Users/Hans/Documents/Arduino/hardware" -tools "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/tools-builder" -tools "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/hardware/tools/avr" -tools "/Users/Hans/Library/Arduino15/packages" -built-in-libraries "/Users/Hans/Downloads/Arduino-1.app/Contents/Java/libraries" -libraries "/Users/Hans/Documents/Arduino/libraries" -fqbn=MightyCore:avr:1284:pinout=standard,variant=modelP,BOD=2v7,clock=16MHz_external -ide-version=10610 -build-path "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/Hans/Downloads/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
Using board '1284' from platform in folder: /Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6
Using core 'MightyCore' from platform in folder: /Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6
Detecting libraries used...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore" "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/variants/standard" "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore" "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/variants/standard" "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore" "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/variants/standard" "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/sketch/Blink.ino.cpp" -o "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Users/Hans/Downloads/Arduino-1.app/Contents/Java/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/cores/MightyCore" "-I/Users/Hans/Library/Arduino15/packages/MightyCore/hardware/avr/1.0.6/variants/standard" "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/sketch/Blink.ino.cpp" -o "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
[... cut ...]
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/Blink.ino.elf" "/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/buildfbb3364bdd983a856e968c32e574d2fe.tmp/Blink.ino.hex"

Sketch uses 1,208 bytes (0%) of program storage space. Maximum is 130,048 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 16,375 bytes for local variables. Maximum is 16,384 bytes.

```

MCUdude, 2016-07-27:
It took some time installing the core from the boards manager. "Installing Tools..." was the message that took the most time

jantje, 2016-07-27:
This explanation is a bit confusing to me. Is there still a problem?

MCUdude, 2016-07-27:
Oh my! The two last post was supposed to be posted some where else. The first post is OK.

jantje, 2016-07-27:
Can you specify a bit more on the "core actually has two pins_arduino.h files."
They can not be in the same folder (at least not when they have the same name.
Maybe it is easier if you provide the json file and the board name so I can try it here.

MCUdude, 2016-07-27:
A link to the core is provided in the first post. The Arduino IDE wants the different pins_arduino.h files to be located in two separate folders. These two files are selected in the boards.txt file. These pins_arduino files can be found [HERE](https://github.com/MCUdude/MightyCore/tree/master/avr/variants). The JSON URL is located in the READBE file

jantje, 2016-07-27:
This has got to do with the fact the variant is changed based on a menu selection.
Apparently this is currently not supported in sloeber.

jantje, 2016-08-20:
This seems to be only when you use core=arduino:arduino
with a variant defined in a menu item.

jantje, 2016-08-20:
Forget the previous remark.
It's getting complicated for just being enable to save some text somewhere.
But it compiles now fine here. You should be able to test tomorrow with the nightly

jantje, 2016-08-20:
I also implemented this change in the hardware spec
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification/_compare/959b5858c7d4229e374e746fce7cb72719e96a7f...bb372fcbcd3caff25a98f7bf6e22cc2af69b0d4e
This solves the fact the tool dependencies is empty.
I don't like it but Arduino core is at call here.

jantje, 2016-09-03:
@MCUdude Is this fixed for you?

MCUdude, 2016-09-05:
I'll test it as soon as possible. How did the JSON tool dependencies issue work out? is it still mandatory, or will the IDE figure it out anyway?

jantje, 2016-09-06:
>I'll test it as soon as possible. How did the JSON tool dependencies issue work out? is it still mandatory, or will the IDE figure it out anyway?

Arduino made a change in the hardware spec. This solves this problem.
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification/_compare/959b5858c7d4229e374e746fce7cb72719e96a7f...bb372fcbcd3caff25a98f7bf6e22cc2af69b0d4e

But you have to reselect the board click apply and ok

closed 2017-01-17 by jantje