Sloeber/arduino-eclipse-plugin #510: Unable to upload code using V3(in mars) or V3.1(neon)

dhrupal93, 2016-08-01:
Hi
I was using eclipse arduino plugin normally on my old laptop.
Now I am installing the same on different PC with WINDOWS 10, but I am not able to upload the program successfully using eclipse arduino plugin using Mars or Neon both.

The build is showing success
[image]

The upload is also showing success, but actually the Arduino MEGA 2560 is not programmed.
What happen is as I click upload, the Arduino MEGA PIN 13 LED flashes, arduino gets reset also. But then again the old program get started.
[image]

I tried to use eclipse version mars and neon both and installed the plugin in them. Also tried the pre-installed package available at http://eclipse.baeyens.it
(Serial monitor of eclipse arduino plugin is working OK)
(Normally arduino is working/programming with Arduino IDE)

jantje, 2016-08-01:
I see you are using the new arvdude tool. Does it work with the arduino IDE?

dhrupal93, 2016-08-01:
I did not get you.
I am using Arduino 1.6.10 separetely and that is working fine
I have downloaded eclipse-Neon eclipse-cpp-neon-R-win32-x86_64 and then added plugin from marketplace. https://marketplace.eclipse.org/content/arduino-eclipse-ide

jantje, 2016-08-01:
Can you compare the arduino upload command to the sloeber upload command

achjiae, 2016-08-06:
Same Problem here.

I use V3.1_win32.2016-08-03_16-31-05 on Win10x64

Sloeber says:

> Starting upload
> using arduino loader
> Starting reset using DTR toggle process
> Toggling DTR
> Continuing to use COM1
> Ending reset
> 
> 
> Launching C:\sloeber\arduinoPlugin\tools\arduino\avrdude\6.3.0-arduino2/bin/avrdude -CC:\sloeber\arduinoPlugin\tools\arduino\avrdude\6.3.0-arduino2/etc/avrdude.conf -patmega328p -carduino -PCOM1 -b115200 -D -Uflash:w:C:\sloeber\workspace\Blink/Release/Blink.hex:i 
> Output:
> 6.3.0-arduino2/bin/avrdude finished
> upload done

When i start the avrdude line in cmd.exe the upload works:

> C:\Windows\system32>C:\sloeber\arduinoPlugin\tools\arduino\avrdude\6.3.0-arduino2/bin/avrdude -CC:\sloeber\arduinoPlugin\tools\arduino\avrdude\6.3.0-arduino2/etc/avrdude.conf -patmega328p -carduino -PCOM1 -b115200 -D -Uflash:w:C:\sloeber\workspace\Blink/Release/Blink.hex:i
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.01s
> 
> avrdude: Device signature = 0x1e950f (probably m328p)
> avrdude: reading input file "C:\sloeber\workspace\Blink/Release/Blink.hex"
> avrdude: writing flash (928 bytes):
> 
> Writing | ################################################## | 100% 0.20s
> 
> avrdude: 928 bytes of flash written
> avrdude: verifying flash memory against C:\sloeber\workspace\Blink/Release/Blink.hex:
> avrdude: load data flash data from input file C:\sloeber\workspace\Blink/Release/Blink.hex:
> avrdude: input file C:\sloeber\workspace\Blink/Release/Blink.hex contains 928 bytes
> avrdude: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 0.16s
> 
> avrdude: verifying ...
> avrdude: 928 bytes of flash verified
> 
> avrdude: safemode: Fuses OK (E:00, H:00, L:00)
> 
> avrdude done.  Thank you.

What can i do to solve this Problem?

jantje, 2016-08-06:
So it looks like trying to reset the board makes it fail :-s

achjiae, 2016-08-06:
So what can I do to fix the problem.
With current Eclipse and the plugin I get the same error.
I tried it with an Arduino Uno and Mega.

As a next step I try with fresh Windows7 and Linux on a VM.

jantje, 2016-08-06:
try arduino avr 1.6.11.
The toolchains have changed in 1.6.12 and I'm pretty sure that that is the root cause.

achjiae, 2016-08-06:
I'm not quite sure what you mean.

I have copied the exe files from the Arduino IDE (1.6.10 and 1.6.0) from
\ Hardware \ tools \ avr \ bin
to
\ ArduinoPlugin \ tools \ arduino \ avr-gcc \ 4.9.2-atmel3.5.3-arduino2 \ bin
\ ArduinoPlugin \ tools \ arduino \ avrdude \ 6.3.0-arduino2 \ bin
Unfortunately the upload still does not work. The Arduino flashes 4 times quickly and 1 second later 1 time quick.

jantje, 2016-08-06:
I strongly advice you not to move files around like that.
Please watch this video. https://www.youtube.com/watch?v=HE5iYxv-B-o&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF
What I'm talking about here starts at 10:04

achjiae, 2016-08-07:
Now i got it and it's working!

For all the other people with the same Problem:

check on Window - Preferences - Arduino - Platforms and Boards- arduino - Arduino AVR Boards - 1.6.11

Thanks a lot.

[image]

jantje, 2016-08-07:
Thanks for confirming this is toolchain related.

achjiae, 2016-08-07:
Additional Information: The error also occurs on fresh Windows 7 32bit. On fresh Linux (Mint18) it works without problems. My Arduino boards using a CH340 Serial USB chipset.

jantje, 2016-08-07:
So it seems to be a windows only problem

jantje, 2016-08-07:
I labelled this boards specific though it is more platform specific

jantje, 2016-08-21:
If you want to use 1.12 please copy the following files from a arduino install into your path or next to the avrdude file
libusb0.dll
msvcp100.dll
msvcr100.dll

jantje, 2016-09-03:
Arduino AVR Toolchain 1.13 reverted back to the old avr dude.

jantje, 2016-10-29:
As the new avrdude no longer has this issue (the dll is part of it) I close this and keep it as a known limitation.
Thanks Arduino core team :-)

closed 2016-10-29 by jantje

lukasadamek, 2017-10-06:
I have still the same problem.
Sloeber:  4.1.0.201705252335
Arduino AVR Board: 1.6.20

The upload is also showing success, but actually the Arduino MEGA 2560 is not programmed.

jantje, 2017-10-06:
please create your own issue as this is not related.
Also provide the projectproperties->arduino screenshot and the console output for the build and upload