Sloeber/arduino-eclipse-plugin #516: 3rd party board not compiling

Steffen-Engel, 2016-08-04:
sloeber works perfectly with Arduino-Board as specified in arduino/Arduino AVR Boards/1.6.12

After installation of adafruit/1.4.9 any of these boards can't compile, due to not found compiler.
in Release/subdir.mk it is

`.ino.cpp.o: ../.ino.cpp
	@echo 'Building file: $<'
	@echo 'Starting C++ compile'
	"/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_ADAFRUIT32U4 -DARDUINO_ARCH_AVR -DUSB_VID=0x239A -DUSB_PID=0x8001 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"32u4 Breakout\""  -I"C:\tmp\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.12\cores\arduino" -I"C:\tmp\sloeber\arduinoPlugin\packages\adafruit\hardware\avr\1.4.9\variants\adafruit32u4" -MMD -MP -MF"$(@:%.o=%.d)" -MT"$@" -D__IN_ECLIPSE__=1 -x c++ "$<" -o "$@"  -Wall
	@echo 'Finished building: $<'
	@echo ' '
`

so there is missing the leading path for /bin/avr-g++

Compared to an arduino board, the path to the compiler is correct:
`
.ino.cpp.o: ../.ino.cpp
	@echo 'Building file: $<'
	@echo 'Starting C++ compile'
	"C:\tmp\sloeber\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\tmp\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.12\cores\arduino" -I"C:\tmp\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.12\variants\leonardo" -MMD -MP -MF"$(@:%.o=%.d)" -MT"$@" -D__IN_ECLIPSE__=1 -x c++ "$<" -o "$@"  -Wall
	@echo 'Finished building: $<'
	@echo ' '
`

on stable Sloeber 3.1, windows 7 64, PATH cleaned from existing sh and make
temporary installation of sloeber in c:\tmp\sloeber

jantje, 2016-08-04:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Steffen-Engel, 2016-08-04:
So busy with other projects, but patron is fine, did it.

jantje, 2016-08-04:
I'm not seeing anything :-s
Could you post here? https://www.patreon.com/posts/i-support-issue-3700469

jantje, 2016-08-04:
Thanks for the support.
I'm trying here right now.

jantje, 2016-08-04:
I can reproduce the problem here. It is related to the fact the tools are not "defined" in the json file.
I'm trying to find a solution.

jantje, 2016-08-05:
```
....
          "version": "1.4.9",
          "architecture": "avr",
          "archiveFileName": "adafruit-avr-1.4.9.tar.bz2",
          "boards": [
            {
              "name": "Adafruit Flora"
            },
.....
          "toolsDependencies": [],
          "name": "Adafruit AVR Boards"
```

This looks like a bug in the json file. Does this work with the Arduino IDE?

Steffen-Engel, 2016-08-05:
thanks for the fast response.

With arduino IDE it works like a charm

I have other issues with other board files, too. Maybe they arise from the same area.

eg 
`
  http://www.leonardomiliani.com/repository/package_leonardomiliani.com_index.json
`
leads to not compiling because of wrong path to compiler:
`
  "C:\tmp\sloeber\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-g++"
`
compared to arduino:
`
  "C:\tmp\sloeber\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++"
` 

while compiler.path is same in both platform.txt:
`
  compiler.path={runtime.tools.avr-gcc.path}/bin/
`

other one is 
`
https://raw.githubusercontent.com/mikaelpatel/Cosa/master/package_cosa_index.json 
`

which doesn't try to compile, just gives the message 'Failed to find core reference Cosa:cosa

All working with standard arduino 'IDE'

jantje, 2016-08-05:
There is a issue with the json description in this area. See https://github.com/arduino/Arduino/issues/5168 for more info.
Until it clears out which direction this is going I do not see any other way out then modifying the json file.

Steffen-Engel, 2016-08-08:
What a pity, so Sloeber is unuseable for me because they will not change the files with Arduino IDE running them correctly...

Steffen-Engel, 2016-08-08:
I am trying to work around with a copy of all necessary files for my boards, but even that does not work.

Copying arduinoPlugin/packages/arduino to arduinoPlugin/packages/testboards gives a selection for the project settings but change to one of those boards does not compile either with the missing path to the compiler...

jantje, 2016-08-08:
I can make a json file for you so the adafruit stuff works :-)
I'll have a look at it tonight (gmt +1)

Steffen-Engel, 2016-08-08:
thanks for that, but not necessary for me. My final goal is a private Board definition where i do collect my boards (some of them selfmade so all those jsons do'nt help me ;-)

Actually I got the private board definition folder (testboards) running by declaring the full path to the compiler in platform.txt:

`
#compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.path=C:\tmp\sloeber\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/
`

Not my favorite but it is a useable workaround.

So it looks like not using the runtime.tools.avr-gcc.path or it is not filed on foreign boards.
i once had the same, when havin arduino1.6.12 installed and added 1.6.11. Variant from 11 could not be compiled due to missing compiler path.
After fiddlign and deactivating 1.6.12 it worked. After reactivating 1.6.12 it still worked.
Can't reproduce at this moment...

jantje, 2016-08-08:
Arduino IDE made lots of assumptions which started causing troubles in the latest release. So they are removing some of them now.
I prefer waiting to see what comes out of it before changing Sloeber.
The problem you are facing is that runtime.tools.avr-gcc.path is defined based on the json content.
Conclusion: I strongly advice you to create a json file for your boards.

jantje, 2016-08-08:
I think it is a good idea to add JANTJE.INSTALLPATH to the environment variables.
Then you could use
compiler.path=${JANTJE.INSTALLPATH}\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/
Which is already a lot better.
But that does not change my mind on advising you to go with a json file.

jantje, 2016-08-21:
I think these issues should be fixed in the nightly. Can you give it a try tomorrow?

jantje, 2016-09-03:
@Steffen-Engel can you give this a test?

Steffen-Engel, 2016-09-05:
Tested a few boards with nightly win32.2016-08-30 , looks VERY good.
Thank you, Jantje.

jantje, 2016-09-05:
Thanks for the feedback

closed 2017-01-17 by jantje