Sloeber/arduino-eclipse-plugin #517: NullPointer Exception on Compilation

scubiedoo, 2016-08-05:
Hi!
I recently updated (from toddler i guess) to sloeber and now get this nullpointer exception whenever i try to compile my project.
What did i do wrong? or is this an issue?
```
java.lang.NullPointerException
	at it.baeyens.arduino.tools.Helpers.setTheEnvironmentVariables(Helpers.java:872)
	at it.baeyens.arduino.listeners.ConfigurationChangeListener.handleEvent(ConfigurationChangeListener.java:47)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.notifyListeners(CProjectDescriptionManager.java:2212)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.fireAboutToApplyEvent(AbstractCProjectDescriptionStorage.java:284)
	at org.eclipse.cdt.internal.core.settings.model.SetCProjectDescriptionOperation.executeOperation(SetCProjectDescriptionOperation.java:66)
	at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
	at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2241)
	at org.eclipse.cdt.internal.core.model.CModelOperation.runOperation(CModelOperation.java:638)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.setProjectDescription(AbstractCProjectDescriptionStorage.java:203)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionStorageManager.setProjectDescription(CProjectDescriptionStorageManager.java:149)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:885)
	at org.eclipse.cdt.managedbuilder.internal.core.BuildSettingsUtil.checkApplyDescription(BuildSettingsUtil.java:211)
	at org.eclipse.cdt.managedbuilder.core.ManagedBuildManager.updateCoreSettings(ManagedBuildManager.java:1574)
	at org.eclipse.cdt.build.internal.core.scannerconfig.CfgDiscoveredPathManager$1.run(CfgDiscoveredPathManager.java:154)
	at org.eclipse.cdt.internal.core.model.BatchOperation.executeOperation(BatchOperation.java:36)
	at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
	at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2241)
	at org.eclipse.cdt.core.model.CoreModel.run(CoreModel.java:1220)
	at org.eclipse.cdt.core.model.CoreModel.run(CoreModel.java:1181)
	at org.eclipse.cdt.build.internal.core.scannerconfig.CfgDiscoveredPathManager.updateCoreSettings(CfgDiscoveredPathManager.java:157)
	at org.eclipse.cdt.build.core.scannerconfig.ScannerConfigBuilder.build(ScannerConfigBuilder.java:125)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:734)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:205)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:245)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:300)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:303)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:262)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:402)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:556)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2241)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:534)
	at org.eclipse.core.internal.resources.Project.build(Project.java:111)
	at it.baeyens.arduino.actions.BuildJobHandler.run(BuildHandler.java:45)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

```

jantje, 2016-08-05:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Apart from that: Check the update instructions as described here: http://eclipse.baeyens.it/how_to.shtml#/n

jrobert-gh, 2016-08-18:
I've also had this happen just recently. It appears that upgrading, then reverting, my version of the Eclipse Arduino 3.1 Product, broke a project. I fixed a number parser and null pointer problems by: 

- Saving the source files from my damaged project;
- Deleting the project and its files from within Eclipse, then closing Eclipse;
- In the ArduinoIDE, creating & compiling a new project with my saved source files. Close the IDE;
- Starting Eclipse and creating a new project with the same name as the new IDE project (I used the same project name throughout this procedure);

The repaired project gave me no more pointer exceptions or Eclipse parsing errors (except for one parsing error that had preceded my breaking the project).

jantje, 2016-09-03:
closing as no sponsor found

closed 2016-09-03 by jantje