Sloeber/arduino-eclipse-plugin #518: Combiner incorrectly interprets spaces in path?

joeinaustin, 2016-08-07:
My build is failing because some part of it fails to interpret the path to one library correctly.

See below partial console output............

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -Os -Wl,--gc-sections -mmcu=atmega328p -o "C:/Users/jburbano/eclipse/workspace/optoread1/Release/optoread1.elf" ./.ino.cpp.o ./Libraries/Wire/src/utility/twi.c.o ./Libraries/Wire/src/Wire.cpp.o ./Libraries/TimerOne/TimerOne.cpp.o ./Libraries/SoftwareSerial/src/SoftwareSerial.cpp.o ./Libraries/NewliquidCrystal/thirdparty\ libraries/SoftI2CMaster_/SoftI2CMaster.cpp.o ./Libraries/NewliquidCrystal/FastIO.cpp.o 

This part is not correct "....thirdparty\ libraries/So......" Note the "\ " where only a space should be. The actual path is "C:\Program Files (x86)\Arduino\libraries\NewliquidCrystal\thirdparty libraries\SoftI2CMaster" Note that only a space is between thridparty and libraries.

I am not sure how to correct this. 

I am running Luna SR2 (4.4.2)
Build id: M20150204-1700 on Win7 x64 with the Arduino plugin

I also posted the above in the Eclipse fourm.  But I think this might be the correct place.

jantje, 2016-08-07:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Apart from that we know spaces are know to give problems. We try to handle them but in general we advice to avoid spaces in path/file names.

joeinaustin, 2016-08-07:
I can remove the space in the actual path but I have no idea how to reference the new path.  I had already created a new directory called "thirdpartylibraries" hoping for a quick fix.  But can't find anywhere to change the ref.  Any ideas?   

On another note, I find it odd that it had no problem with the spaces at the beginning of the path.

joeinaustin, 2016-08-07:
BTW, I did not create any of these directories or path references.  I only installed the Arduino plugin and added the library called NewliquidCrystal.  I believe it is that library that references the path with the space.  This is the same library that works w/o issue in the standard Arduino.cc IDE.  That IDE must properly interpret the path so it seems it is something in Eclipse or the Arduino plugin that is causing my issue.

joeinaustin, 2016-08-07:
The bug remains but I found a simple workaround.  I started Eclipse as an administrator to allow re-write of the "program files" directories.  Then I simply used Eclipse to "rename" the library to someting w/o a space in project explorer.

I believe the creator of the library tested it in the Arduino.cc IDE and therefore was unaware that the spaces are not properly handled by Eclipse.  So there is nothing to prevent this from happening again.

closed 2016-10-27 by jantje