Sloeber/arduino-eclipse-plugin #519: Missing message: doYouWantToBuildBeforeUpload

EdwinCroissant, 2016-08-08:
One picture can say more then a thousand words :)

[image]

Windows 10 and the nightly.

jantje, 2016-08-08:
Thanks
Should be fixed tomorrow :-)

EdwinCroissant, 2016-08-09:
It's fixed, thank you!
Did the nag screen became more frequent in the process?

jantje, 2016-08-09:
Thanks for confirming it works.
I'm working on a way for my Patron to not have to see this anymore. Once that works I'll probably increase the nag frequency.
But I have seen it quite a lot as well lately, So I may have left some debugging parameters in the code when debugging. I should take a look.

jrobert-gh, 2016-08-22:
[Semi-related]: Where can I find/change this preference?

jantje, 2016-08-22:
I just updated the patreon awards to include a nag-less experience :-)

jrobert-gh, 2016-08-22:
Excellent! 

(But if that was a reply to my few-minutes-ago question, I don't understand. I'm looking for how to turn-on/off auto build-before-upload. I turned it off in a pop-up dialog some time ago. Now I can't find it to turn it on again.

jantje, 2016-08-22:
The options are in windows->preferences->arduino

jantje, 2016-12-08:
closing as this was a nightly regression.

closed 2016-12-08 by jantje