Sloeber/arduino-eclipse-plugin #523: Options errors in Arduino Assembler

llemtt, 2016-08-18:
Arduino Assembler dependency options in ${FLAGS} generated by DefaultGCCDependencyCalculator2Commands are wrong and cause errors with some toolchains assemblers.

RedBearLab boards package for instance uses arm-none-eabi-as assembler that doesn't support the same dependency options that arm-none-eabi-gcc does. (-MD only, not -MF _MT...)

The arm-none-eabi-as assembler doesn't seem to support also the -D option.

As a workaround I had to remove ${FLAGS} and -D__IN_ECLIPSE__=1 from the build command line.

jantje, 2016-08-18:
@llemtt 
Thanks for the input. After I re-enabled assembly I never got confirmation it actually worked.
If I understand it correctly it works but not for all assemblers.
What I'm trying to say: "I don't know sh** about assembly, so I'll need your help to get it fixed properly."
Are you saying you removed `${FLAGS} -D__IN_ECLIPSE__=1 ` from the command line pattern? In the setting below?
[image]

How does this relate to DefaultGCCDependencyCalculator2Commands?

llemtt, 2016-08-18:
Yes I removed from that setting!

Assembler for other boards, such as Arduino Uno or Arduino Due, works correctly because it's invoked via the C compiler using "... arm-none-eabi-gcc -c -g -x **assembler-with-cpp** ..." instead of invoking arm-none-eabi-as directly.

Still don't know how other third party boards behave.

DefaultGCCDependencyCalculator2Commands is invoked by

`private String[] addDependencyOptions(IManagedDependencyCommands depCommands, String[] flags) {
		String[] depOptions = depCommands.getDependencyCommandOptions();`

in ArduinoGnuMakefileGenerator.java, considering that

`public class ArduinoDependencyCalculatorCommands extends DefaultGCCDependencyCalculator2Commands`

getDependencyCommandOptions() in DefaultGCCDependencyCalculator2Commands returns "hard coded" dependency options valid only for the C/C++ compiler. The result is pushed into FLAGS.

jantje, 2016-08-18:
If I understand this correctly changing 
ArduinoDependencyCalculatorCommands 
If the language is assembly and the commands contains "assembler-with-cpp" do as now and add :+1: `" -DIN_ECLIPSE=1"`
else return nothing should 
and removing the ` -DIN_ECLIPSE=1` from the command should do the trick.

Next question... Do we have access to the command and language in ArduinoDependencyCalculatorCommands

llemtt, 2016-08-18:
Yes it should work.

addDependencyOptions I see is called from inside addRuleForSource at the beginning of which the "tool" object is retrieved from Eclipse. From the "tool" object the command and language should be available but the object has to be passed as a parameter to addDependencyOptions.

Inside addDependencyOptions  I think it would be better to detect which tool is actually used and generate the correct options:

if (tool is a "managed build compatible" C/C++ compiler [like gcc/g++, I don't know how one could correctly detect it]) then "do as now" [i.e. invoke getDependencyCommandOptions]

else if (tool is gnu assembler [like arm-none-eabi-as]) then push "--MD\"$(@:%.o=%.d)\"" into FLAGS and remove -D__IN_ECLIPSE__=1 and eventually other "-D" options

else if (tool is...) then...
else push nothing and remove -D__IN_ECLIPSE__=1 and eventually other "-D" options

However it would be better if all of this was managed correctly inside Eclipse so that you can just invoke getDependencyCommandOptions!!

jantje, 2016-08-18:
I'm currently working on some major refactoring to get junit testing.
I thought this was going to be finished today but it seems I forgot a part :-(
I'll handle this issue after that.

jantje, 2016-08-20:
Can you provide a link to the json file and a board specification and a asm file so I can reproduce the issue here?
Best would be that the asm is a example of a library. This way I should be able to integrate it in mu -newly created- unit tests.

llemtt, 2016-08-22:
Here is the link

https://redbearlab.github.io/arduino/package_redbearlab_index.json

You can build any of the examples and the file

core/core/startup_NRF51822.S

will be assembled.

llemtt, 2016-08-24:
Oh I see, that's a more subtle bug I forgot I had to fix..

It's caused by a wrong order in JANTE.OBJCOPY variable

${A.RECIPE.OBJCOPY.HEX.PATTERN}
	${A.RECIPE.OBJCOPY.HEX1.PATTERN}
	${A.RECIPE.OBJCOPY.HEX2.PATTERN}
	${A.RECIPE.OBJCOPY.EEP.PATTERN}

The EEP.PATTERN must be executed before the HEX2.PATTERN because the latter depends on the .bin file created by the former.

I have overridden JANTE.OBJCOPY with the following:

${A.RECIPE.OBJCOPY.HEX.PATTERN}
	${A.RECIPE.OBJCOPY.HEX1.PATTERN}
	${A.RECIPE.OBJCOPY.EEP.PATTERN}
	${A.RECIPE.OBJCOPY.HEX2.PATTERN}

The bug is subtle because once you do a second build the error disappear but then you upload the .hex of the previous build!!

jantje, 2016-08-24:
I order the list now. So it becomes
${A.RECIPE.OBJCOPY.EEP.PATTERN}
${A.RECIPE.OBJCOPY.HEX.PATTERN}
${A.RECIPE.OBJCOPY.HEX1.PATTERN}
${A.RECIPE.OBJCOPY.HEX2.PATTERN}
That seems to work

jantje, 2016-09-03:
@llemtt Does this work for you now?

llemtt, 2016-09-05:
Don't know how to properly test the fix, which version should I use?

rlogiacco, 2016-09-05:
I believe you should update to the latest nightly build to get the most recent patches...

llemtt, 2016-09-05:
Tested with latest nightly build it works!

jantje, 2016-09-05:
txs for testing. I reopened as I keep issues open untill they are part of the stable

closed 2017-01-17 by jantje