Sloeber/arduino-eclipse-plugin #524: Automatically switch console-window tabs on Verify or Upload

jrobert-gh, 2016-08-18:
It takes lot of tab changing to watch my compiles, and to watch the newly uploaded program's terminal output. 
It would be helpful to auto-switch to the console tab on a Verify or Upload, and switch to the terminal output tab (or a selected one, if there are several) when an upload ends successfully.

jantje, 2016-08-19:
When a build is done the console should be activated. I'm not sure why it doesn't anymore. I'll have a look.
Activating another window after an action is not done in eclipse.
Are you aware that you can put them next to each other so you can see both of them at the same time?

jrobert-gh, 2016-08-20:
> put them next to each other

Too bad - that is a handy feature of the Arduino IDE. Making both visible would be an ok work-around.

jantje, 2016-08-20:
I'll have a look and see whether it is feasible to -optionally- activate the serial window after a upload.
But it is low priority as making them both visible is possible right now and is the "preferred eclipse way"

jrobert-gh, 2016-08-20:
I just noticed - the console _does_ get activated on Upload, just not on
Verify.

jantje, 2016-08-21:
Implemented the console switch for upload and build. Should be available after 06:00 GMT on the nightly

jrobert-gh, 2016-08-22:
Confirmed - thanks.

closed 2017-01-17 by jantje