Sloeber/arduino-eclipse-plugin #526: Adafruit pro trinket fails to compile

jantje, 2016-08-21:
The linkt to the variant folder goes to 
[eclipse install]\arduinoPlugin\packages\arduino\hardware\avr\1.6.12\platform.txt\variants\eightanaloginputs\eightanaloginputs 
but it should be
[eclipse install]\arduinoPlugin\packages\arduino\hardware\avr\1.6.12\variants\eightanaloginputs

Once this is changed it compiles fine.

rlogiacco, 2016-09-02:
@jantie how do you pick the version number to use in the above path? There might be an error there.. are you picking the latest available?

jantje, 2016-09-02:
The version number comes from the boards.txt file here.
The error in this particular situation was that the platform.txt file was added and the variants folder was added 2 times (at different locations)

closed 2017-01-17 by jantje