Sloeber/arduino-eclipse-plugin #529: avrdude: Yikes! Invalid device signature.

hassanbot, 2016-08-24:
I can't get my code to upload to the Arduino, even though the compiling and setup seems fine. Uploading works fine with the Arduino IDE, so there should be nothing wrong with the connections. The console gives the following output:
```
Starting upload
using arduino loader

Launching /home/martin/Developer/tools/eclipse/sloeber/arduinoPlugin/tools/arduino/avrdude/6.0.1-arduino5/bin/avrdude -C/home/martin/Developer/tools/eclipse/sloeber/arduinoPlugin/tools/arduino/avrdude/6.0.1-arduino5/etc/avrdude.conf -patmega328p -cstk500v1 -P/dev/ttyACM1 -Uflash:w:/home/martin/Developer/tools/eclipse/sloeber/workspace/kokoko/Release/kokoko.hex:i 
Output:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

avrdude finished
upload done
```

OS: Ubuntu 16.04
Board: Arduino Uno
I'm using the same settings for port and programmer (/dev/ttyACM1 and AVR ISP) that works with the Arduino IDE. Not sure how to continue the debugging from here.

jantje, 2016-08-24:
Why not use default programmer?

hassanbot, 2016-08-25:
Yeah, that worked, don't know why I didn't try that. Thanks!

closed 2016-08-25 by hassanbot