Sloeber/arduino-eclipse-plugin #531: what is the official way to link projects?

jonbarril, 2016-09-01:
I checked FAQ and other places.  I've poked around Arduino plugin and Eclipse settings. The most obvious way to link projects is via Eclipse "project references", but that does not seem to work. The best way I have found is to use "Arduino->Add a source folder to the selected project", and link to the project folder.  Although this works it does not seem to be consistent with the "eclipse way".

What is the official way to link Arduino projects in eclipse?

Now for the bug/improvement....I suggest that Eclipse "project references" be the way to do this, as this is the way to do it in Eclipse when doing other forms of development.

jantje, 2016-09-01:
It is unclear to me what you are asking

jonbarril, 2016-09-02:
Say I have an eclipse workspace with two Arduino projects. Project A 
depends on project B (e.g. a project A header file includes a project B 
header file). How do I configure project A so that it compiles and uses 
project B?

Hope this is clearer.

jantje, 2016-09-02:
With project references. Why do you think this doesn't work?

jonbarril, 2016-09-02:
For the second time, I recently created a new project and wanted to 
reference another project where I was building some utilities. The 
include file statement to the referenced project include file could no 
be resolved by the IDE, and trying to compile and run the project 
produced an error message about the referenced include file not being found.

Playing around with the eclipse and arduino settings , and looking at 
what allowed another project to use the same utilities in the referenced 
project, the only way I could get it to work was to use "Arduino->add a 
source folder to the selected project". This created a link to the 
referenced project sources, and allowed the target project to compile 
and run.

Using Eclipse "referenced projects" clearly was not sufficient to allow 
this to happen (tried opening and closing the projects, as well as 
restarting Eclipse to help it along).  Should it be?

--jon

jantje, 2016-09-03:
I still don't understand what you are trying to do. But from the looks of it; I think it is more cdt related then sloeber related.
Sloeber projects are CDT managed build projects. CDT supports project dependencies and V1 of the arduino eclipse plugin used it to have a library project for the board and a sketch project.
That worked fine but was removed due to Arduino requirements not fitting this picture.

closed 2016-09-03 by jantje

jonbarril, 2016-09-09:
It seems the confusion arises from eclipse having two rays to "reference projects". One is Properties->Project references. The other is Properties->C/C++ General->Paths and Symbols->References.

For other development I have used the former. It seems in CDT (and the Arduino Plugin) only the latter works -- hence the confusion.

jonbarril, 2016-09-09:
In each project I have an .ino file that I use to test the stuff in that project. However, if I reference one project from another it complains about multiple setup() and loop() definitions. Is there someway to designate a given ino as the "main" one?