Sloeber/arduino-eclipse-plugin #532: Platforms and Boards displays incorrectly

rlogiacco, 2016-09-01:
The way the "Platforms and Boards" list is displayed sorts the versions using string comparison, leading to a wrong sorting:
[image]

Solution is to parse the version number.
Also checking/unchecking the parent checkbox is (for a good reason) impossible: the visualization should be changed similarly to the libraries manager.

jantje, 2016-09-02:
When you are at it. can you also order the platforms?
The checkboxes are already mentioned in #398. Shall I close that as a duplicate of this one?
Note that the Manager class contains a Static compareVersions method taking 2 string.
I get confused quite often within this dialogbox (when there are plenty of boards installed) which is caused by the fact the package json file is not visible. This makes it possible there are 2 entries with the same name.

rlogiacco, 2016-09-02:
Regarding #398 I've assigned that too, no need to close as duplicate for now.
Regarding the Manager class I intend to refactor the whole thing to have one version parsing shared with the libraries, which is working.

jantje, 2016-09-03:
OK, Just don't undo my split from gui and controller in sloeber.io :-) and test the junit script :-)

rlogiacco, 2016-09-03:
@jantje ` which is caused by the fact the package json file is not visible`
Are you suggesting we should also show the package source file or just drop duplicates and pick only one of the two?

@jantje `can you also order the platforms?`
Any suggestion regarding the best ordering or just alpha (which, I believe, is already in place)?

jantje, 2016-09-03:
you can not drop duplicates as it is only the platform name that is duplicate. The boards may/are different

With me the platforms are not ordered case insensitive
[image]
- )

I would like to have the jason file visible one way or another. Maybe as a filter or just as part of the tree

jantje, 2016-10-07:
I have handled all the issues here.
Based on your library work :-)
[image]

closed 2017-01-17 by jantje