Sloeber/arduino-eclipse-plugin #535: icns damaged?

cyberreefguru, 2016-09-13:
I recently installed the latest stable release of Sloeber on my Mac.  When I moved the application to the dock, the dock icon shows up as a generic icon, not the Sloeber icon.  When I use the get info on the application, it shows the icon in the top left, but under preview is shows the generic icon.  When I try to view the icon in Contents/Resources finder says the icon is damaged. 

I copied the icon from GitHub and pasted it into the application info and it works fine. 

I'll try to debug more when I get a chance.

Possible operator error but I wanted to report it anyway.

Mac OS 10.11.6
Sloeber Version V3.1_mac64.2016-08-03_16-31-05.tar.gz

jantje, 2016-09-13:
:-) nice to see how Sloeber is being used. If you find the issue please create a pull request.

jantje, 2017-01-05:
@mjmeijer Can you check you have the same problem?

jantje, 2017-01-05:
@mjmeijer 
Thanks for verifying
@cyberreefguru 
Are you ok to close this issue?

closed 2017-01-28 by jantje