Sloeber/arduino-eclipse-plugin #536: Usage of enum declaration used in prototype declaration in .ino

HolgerMeiners, 2016-09-18:
When using a enum declaration in the .ino that is used in the prototypes that are directly created in the .ino file the compilation fails. This is due to the fact that the code generator copies the prototypes to the .ino.cpp however the the enum is only declared in the later included .ino file.

jantje, 2016-09-18:
This is a known limitation. I have just created a post for my patrons to explain this in full extend and how to deal with it.
https://www.patreon.com/posts/defines-before-6795548
Become a patron and read.

closed 2016-09-18 by jantje

jantje, 2016-09-18:
Rereading this issue I see it is related but not the same thing as I explain in the post mentioned above.
However this is a limitation which I can't resolve.
If you want to fix this using .ino files define you enum in a header file. (Use include guards)

HolgerMeiners, 2016-09-19:
Well, I was wondering as well why the prototypes are copied to the .ino.cpp even though they are defined in the .ino which is included just below.

jantje, 2016-09-19:
That is what .ino is all about. If you do not need that use .cpp :-)