Sloeber/arduino-eclipse-plugin #537: USBtinyisp (ATtiny) programmer generates a null pointer exception

lakidd, 2016-09-22:
As per  patreon.com/posts/3700469
I'm using the ATTinyCore here
https://github.com/SpenceKonde/ATTinyCore
And it lets me select Upload Protocol to select the programmer.
USBasp (ATtiny) works fine but USBtinyisp (ATtiny) generates a NullPointerException
[image]
I can't seem to find how to generate a stack trace ?

jantje, 2016-09-22:
I can reproduce the error here with 1.1.0 but there is no problem with 1.1.1.
So I guess there is something "wrong" in 1.1.0. A nicer message should be given though.
As we're in refactoring right now I can't work on that error right now.

If you do not see version 1.1.1 goto windows->preferences>arduino locations and check update local json files (including library) select ok, wait a while and try again.

lakidd, 2016-09-22:
Yep, that fixes it for me too. Thanks, I'll use 1.1.1

jantje, 2016-09-23:
txs for the feedback :-)

jantje, 2016-10-07:
I replaced the null pointer error with a nicer message.

lakidd, 2016-10-08:
Awesome thanks Jantje !

Sent from my phone so please excuse brevity and typos

closed 2017-01-17 by jantje