Sloeber/arduino-eclipse-plugin #539: Build fails with maven

jantje, 2016-09-29:
This is what jenkins consistently reports
```
Started by timer
Building in workspace /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace
 > /Apps/bin/git rev-parse --is-inside-work-tree # timeout=10
Fetching changes from the remote Git repository
 > /Apps/bin/git config remote.origin.url https://jantje@github.com/jantje/arduino-eclipse-plugin.git # timeout=10
Fetching upstream changes from https://jantje@github.com/jantje/arduino-eclipse-plugin.git
 > /Apps/bin/git --version # timeout=10
 > /Apps/bin/git -c core.askpass=true fetch --tags --progress https://jantje@github.com/jantje/arduino-eclipse-plugin.git +refs/heads/*:refs/remotes/origin/*
 > /Apps/bin/git rev-parse refs/remotes/origin/master^{commit} # timeout=10
 > /Apps/bin/git rev-parse refs/remotes/origin/origin/master^{commit} # timeout=10
Checking out Revision 189268816c40ef46c3ba356a5a11a6f9a6be61b1 (refs/remotes/origin/master)
 > /Apps/bin/git config core.sparsecheckout # timeout=10
 > /Apps/bin/git checkout -f 189268816c40ef46c3ba356a5a11a6f9a6be61b1
 > /Apps/bin/git rev-list 189268816c40ef46c3ba356a5a11a6f9a6be61b1 # timeout=10
Run condition [Execute Shell] enabling prebuild for step [[Send build artifacts over FTP]]
[workspace] $ /bin/ash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/hudson3483201114776875503.sh
Parsing POMs
Failed to transfer Could not find artifact com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in central (http://repo.maven.apache.org/maven2)
ERROR: Failed to parse POMs
org.apache.maven.project.ProjectBuildingException: Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM: Could not find artifact com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in central (http://repo.maven.apache.org/maven2) and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Could not find artifact com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:3.1.0-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 10
[FATAL] Non-resolvable parent POM: Could not find artifact com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:3.1.0-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 4, column 10
[FATAL] Non-resolvable parent POM: Could not find artifact com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:3.1.0-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11

	at org.apache.maven.project.DefaultProjectBuilder.build(DefaultProjectBuilder.java:364)
	at hudson.maven.MavenEmbedder.buildProjects(MavenEmbedder.java:361)
	at hudson.maven.MavenEmbedder.readProjects(MavenEmbedder.java:331)
	at hudson.maven.MavenModuleSetBuild$PomParser.invoke(MavenModuleSetBuild.java:1301)
	at hudson.maven.MavenModuleSetBuild$PomParser.invoke(MavenModuleSetBuild.java:1098)
	at hudson.FilePath.act(FilePath.java:990)
	at hudson.FilePath.act(FilePath.java:968)
	at hudson.maven.MavenModuleSetBuild$MavenModuleSetBuildExecution.parsePoms(MavenModuleSetBuild.java:960)
	at hudson.maven.MavenModuleSetBuild$MavenModuleSetBuildExecution.doRun(MavenModuleSetBuild.java:679)
	at hudson.model.AbstractBuild$AbstractBuildExecution.run(AbstractBuild.java:534)
	at hudson.model.Run.execute(Run.java:1738)
	at hudson.maven.MavenModuleSetBuild.run(MavenModuleSetBuild.java:531)
	at hudson.model.ResourceController.execute(ResourceController.java:98)
	at hudson.model.Executor.run(Executor.java:410)
[workspace] $ /bin/ash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/hudson4434096625650381180.sh
cd: 3: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86
tar: sloeber: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
cd: 5: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86_64
tar: sloeber: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
cd: 7: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86_64
tar: sloeber: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
cd: 9: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/win32/win32/x86
tar: sloeber: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
cd: 11: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/macosx/cocoa/x86_64
tar: sloeber.app: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
[workspace] $ /bin/bash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/CommandInterpreterCondition914592292736802771.sh

No change
Run condition [Execute Shell] preventing perform for step [[Send build artifacts over FTP]]
Finished: FAILURE
```

Help from the guru's is appreciated.

rlogiacco, 2016-10-01:
So, here we are creating a mess with Maven `groupId` and `artifactId`: the former should be `io.sloeber` and the latter should identify the module _within_ the Sloeber world...
I'm not sure who/what `com.github.brodykenrick` is, but it should be replaced; also you want only `parent` or `api` as artifact ids... not that is the source of the problem, but it's against Maven conventions.

With regards to the specific issue, we are referencing two versions of the exact same project/module: `4.0.0.qualifier` and `3.1.0-SNAPSHOT`... that's where I would point my finger at

jantje, 2016-10-01:
>With regards to the specific issue, we are referencing two versions of the exact same project/module: 4.0.0.qualifier and 3.1.0-SNAPSHOT

:-) A search revealed remaining 3.1.0-SNAPSHOT references. Is this necessary to specify all these versions everywhere? Can't we assume the "latest"?

@brodykenrick Set this up a long time ago.
If I remember correctly com.github.brodykenrick was used because of the travis integration. brodykenrick informed me about this and proposed that this would be replaced by something Jantje but that needed accounts and other actions on my behalf that I didn't understand (like this whole maven implementation) so I didn't do this.
I remember @wimjongman having a discussion about this some years ago to change it but then we didn't act upon it (for as far as I can recall the same problems with external tool integrations)
See this nice documentation here https://github.com/jantje/arduino-eclipse-plugin/blob/master/readme_travis_deploy.md for more details.

jantje, 2016-10-01:
I get this error now :-(
Started by user anonymous
Building in workspace /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace
 > /Apps/bin/git rev-parse --is-inside-work-tree # timeout=10
Fetching changes from the remote Git repository
 > /Apps/bin/git config remote.origin.url https://jantje@github.com/jantje/arduino-eclipse-plugin.git # timeout=10
Fetching upstream changes from https://jantje@github.com/jantje/arduino-eclipse-plugin.git
 > /Apps/bin/git --version # timeout=10
 > /Apps/bin/git -c core.askpass=true fetch --tags --progress https://jantje@github.com/jantje/arduino-eclipse-plugin.git +refs/heads/*:refs/remotes/origin/*
 > /Apps/bin/git rev-parse refs/remotes/origin/master^{commit} # timeout=10
 > /Apps/bin/git rev-parse refs/remotes/origin/origin/master^{commit} # timeout=10
Checking out Revision cd30130e0c3cc9e930be19b969760d11a05a3bd7 (refs/remotes/origin/master)
 > /Apps/bin/git config core.sparsecheckout # timeout=10
 > /Apps/bin/git checkout -f cd30130e0c3cc9e930be19b969760d11a05a3bd7
 > /Apps/bin/git rev-list 30320bfa9196f0684a5ece0a6b7b86b7c6ec5b73 # timeout=10
Run condition [Execute Shell] enabling prebuild for step [[Send build artifacts over FTP]]
[workspace] $ /bin/ash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/hudson2504021334157979815.sh
Parsing POMs

ERROR: Failed to parse POMs
org.apache.maven.project.ProjectBuildingException: Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 10
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 10
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11
[FATAL] Non-resolvable parent POM: Failure to find com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:pom:4.0.0.qualifier in http://repo.maven.apache.org/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of central has elapsed or updates are forced and 'parent.relativePath' points at wrong local POM @ line 4, column 11

	at org.apache.maven.project.DefaultProjectBuilder.build(DefaultProjectBuilder.java:364)
	at hudson.maven.MavenEmbedder.buildProjects(MavenEmbedder.java:361)
	at hudson.maven.MavenEmbedder.readProjects(MavenEmbedder.java:331)
	at hudson.maven.MavenModuleSetBuild$PomParser.invoke(MavenModuleSetBuild.java:1301)
	at hudson.maven.MavenModuleSetBuild$PomParser.invoke(MavenModuleSetBuild.java:1098)
	at hudson.FilePath.act(FilePath.java:990)
	at hudson.FilePath.act(FilePath.java:968)
	at hudson.maven.MavenModuleSetBuild$MavenModuleSetBuildExecution.parsePoms(MavenModuleSetBuild.java:960)
	at hudson.maven.MavenModuleSetBuild$MavenModuleSetBuildExecution.doRun(MavenModuleSetBuild.java:679)
	at hudson.model.AbstractBuild$AbstractBuildExecution.run(AbstractBuild.java:534)
	at hudson.model.Run.execute(Run.java:1738)
	at hudson.maven.MavenModuleSetBuild.run(MavenModuleSetBuild.java:531)
	at hudson.model.ResourceController.execute(ResourceController.java:98)
	at hudson.model.Executor.run(Executor.java:410)
[workspace] $ /bin/ash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/hudson4429780751318249353.sh
cd: 3: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/linux/gtk/x86
[... cut ...]
cd: 11: can't cd to /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/it.baeyens.arduino.product/target/products/it.baeyens.arduino.product/macosx/cocoa/x86_64
tar: sloeber.app: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors

[workspace] $ /bin/bash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/CommandInterpreterCondition3363959834625674057.sh
commit cd30130e0c3cc9e930be19b969760d11a05a3bd7
change
Run condition [Execute Shell] enabling perform for step [[Send build artifacts over FTP]]
FTP: Current build result is [FAILURE], not going to run.
Finished: FAILURE

jantje, 2016-10-02:
It is rolling again. I also needed to change the upload on jenkins due to the name changes

closed 2016-10-02 by jantje