Sloeber/arduino-eclipse-plugin #540: Update to V3 fails to install completely

DrObscure, 2016-10-01:
After the update to V3 my Arduino icons, projects, etc have all disappeared, and
I now get the following error on update:
Cannot complete the install because one or more required items could not be found.
  Software being installed: Arduino Eclipse Extensions 3.1.0.201608012005 (it.baeyens.arduino.feature.feature.group 3.1.0.201608012005)
  Missing requirement: Monitor 3.1.0.201608012005 (it.baeyens.arduino.monitor 3.1.0.201608012005) requires 'bundle org.eclipse.ui.views 3.8.0' but it could not be found
  Cannot satisfy dependency:
    From: Arduino Eclipse Extensions 3.1.0.201608012005 (it.baeyens.arduino.feature.feature.group 3.1.0.201608012005)
    To: it.baeyens.arduino.monitor [3.1.0.201608012005]

What did I miss in the config?

jantje, 2016-10-01:
I assume you use a existing eclipse install and add Sloeber to that one.
Given this assumption which eclipse version do you use?

DrObscure, 2016-10-01:
The eclipse version is 
Version: Juno Service Release 2
Build id: 20130225-0426

I upgraded my Arduino toolchain from v1.5.8 to v1.6.6 to accommodate the RFDuino hardware..
I had been running v2.2.1 of Sloeber, and tried to upgrade to v3

I am running on Windoze 8.0

jantje, 2016-10-02:
As you upgrade from Sloeber V2.2.1 to V3.1 it is very strongly recommended to watch the what is new and noteworthy video's.
https://www.youtube.com/watch?v=HE5iYxv-B-o&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF
I would not be surprised V3.1 no longer works with Juno. 
Please watch the new and noteworthy video's and try with the product http://eclipse.baeyens.it/stable.php?OS=Windows
And keep us posted on progress

DrObscure, 2016-10-03:
I have watched the videos, and like very much the features of V3.x.. but, if juno will not accommodate the new version then I may have to finally upgrade my eclipse environment to luna.. and all the other related plugins I am currently using with it.. :)  until then I was going to try and install V2.4 in juno, if you think that's still a viable mix..  However, I was unable to locate the bundle in the archive.. and http://www.baeyens.it/eclipse/V2 doesn't seem to work at the moment for 'new software' adds..

jantje, 2016-10-03:
Lots of questions.
I remember we decided to do something which would drop juno support. I however can not recall when what.
For other readers with: The same thing happened with xp in V3
I agree that upgrading eclipse with all related plugins is a pain. A plugin called oomph is created to solve this pain. However I hate the oomph recorder and I advice it to turn it off with Sloeber as will fail.
You can have as many eclipse instances on your system as your disk will hold. Sharing workspaces between different versions of eclipse (neon, mars, luna, kepler, juno,..) often does not work.

By the way juno is already 4 year old. Why do you want to upgrade to luna and not neon?

> Can I also install an older Arduino like V1.6.6 with the V3 product? (I'm stuck with V1.6.6 at the moment unfortunately)

V3 works with Arduino toolchain versions and not with Arduino IDE versions.
Default the latest toolchain version will be installed but nothing stops you from adding/replacing with the toolchain of your choice.

Please consider showing your appreciation for support https://www.patreon.com/jantje

DrObscure, 2016-10-03:
I was considering Luna or Mars because that's what the V3 plugin is stated to support.. If the V3 is compatible with Neon, then yes, I will take the plunge and go for the latest version..  If all goes well I intend to remove Juno and have a single eclipse install workspace..  

Is there info on how to replace the arduino tool chain to an older version?

jantje, 2016-10-03:
V3 product comes with neon included.

>I was considering Luna or Mars because that's what the V3 plugin is stated to support..

Where did you find that? So I can fix it.

>Is there info on how to replace the arduino tool chain to an older version?

In part 3 I talk about support for different versions of toolchains. It is a simple click in windows->preferences->arduino->boards and platforms. You do have to wait for the toolchain download before you can use it.

>(PS I did just become a Patreon.. thanks again)

Thanks :-))))

DrObscure, 2016-10-03:
The Luna/Mars eclipse version was implied from the 'Build Options' section of this depository on the main page. I'm in the process of installing and config to neon..  only minor system issues..

DrObscure, 2016-10-04:
Ok. neon is up and the V3 plugin is installed, correctly I believe..
Eclipse does complain about an environment variable $HOME for Git.. not sure what it wants here..
Also, has anyone installed the RFDuino libs in teh plugin yet??
Where would I put the RFDuino package tree, under the ...neon\arduinoPlugin\packages\RFDuino ??

DrObscure, 2016-10-04:
It seems I have answered my own questions :) (added the HOME var to env for my GitHub)
I was able to use the package listed for RFduino and 1.6.6, added it to the library list and it loaded up just as expected with the boards, etc added.. 
I am still having some issues with make and the build process but need to investigate it a bit more before bothering you with it.. 
--ttfn, chane

DrObscure, 2016-10-04:
Build issues were due to several copies of sh.exe that had to be contained to avoid going astray during the make..  
All seems to be working great, as expected :) 
Thanks again.. and time for a beer or three

closed 2016-10-04 by DrObscure

jantje, 2016-10-04:
thanks for the feedback and support.
You can have libraries anywhere on your system. Please do not change the content of [eclipseinstall]/arduinoPlugin. Point to a location in windows->preferences->arduino for libraries and additional hardware.

alexpt10, 2016-10-17:
hi, i solve this problema, try change repository to:
http://www.baeyens.it/eclipse/update

i hope help us.