Sloeber/arduino-eclipse-plugin #541: Build fails: Invalid .ino.cpp file created

mmikkone, 2016-10-03:
Arduino Eclipse makes a funny looking file called .ino.cpp and then tries to compile it when I hit verify button. Here all the classes are missing, for instance:

[image]

This is what my .ino file looks like:

[image]

I don't really know what to say more since I'm completely new with Arduino Eclipse. Just ask me and I'll be more specific.

jantje, 2016-10-03:
@mmikkone 
The .ino.cpp file is a file Sloeber uses to enable .ino support.
From the looks of what I see  there is a problem with the .ino.cpp file when there are classes in your .ino files.
The problem is that method definitions for class methods are created in the ino.cpp file and they should not.
We are currently reorganizing the code and things are not yet stable. So I think your best way forward is a workaround.
I see 2 workarounds:
1) rename your .ino to .cpp (it might be nessesary to add function declarations like the .ino.cpp does)
2) move all classes to their own .h .cpp file.

mmikkone, 2016-10-03:
Thank you! I tried no. 2 and it worked!

jantje, 2016-10-03:
:-)

martinmikula, 2016-11-25:
I have another problem with .ino.cpp file, when i tried to use library "MySensors", the content of the file is wrong. This library needs some configuration defines before import header files:

> #include "Arduino.h"
> 
> // Enable debug prints to serial monitor
> #define MY_DEBUG
> 
> // Enable and select radio type attached
> #define MY_RADIO_NRF24
> 
> #include <MySensors.h>
> 
> int BATTERY_SENSE_PIN = A0;  // select the input pin for the battery sense point
> 
> unsigned long SLEEP_TIME = 900000;  // sleep time between reads (seconds * 1000 milliseconds)
> int oldBatteryPcnt = 0;
> 
> void setup()
> {
> 

 but the imports are in the begining of the file and the defines aren't.

> #ifdef __IN_ECLIPSE__
> //This is a automatic generated file
> //Please do not modify this file
> //If you touch this file your change will be overwritten during the next build
> //This file has been generated on 2016-11-25 13:43:07
> 
> #include "Arduino.h"
> #include "Arduino.h"
> #include <MySensors.h> <-- problem, the file needs defines...
> void setup() ;
> void presentation() ;
> void loop() ;
> 
> #include "mySensors.ino"
> 
> #endif

When i look into arduino sketch cpp file (temp/sketch/sketch_<project>.ino.cpp) (something similar as ino.cpp), there is no include of ino file, but they rather copy the "ino" file and add function declarations after includes... Maybe it wold be better to use the same approach to keep the compatibility...

jantje, 2016-11-25:
@martinmikula 
This is a known issue ( #478) that is unrelated to this issue.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-01-17 by jantje