Sloeber/arduino-eclipse-plugin #542: embedded EspExceptionDecoder possible?

moelski, 2016-10-07:
Hi !

The EspExceptionDecoder has helped alot in the past. So it would be great if that could be implemented in the this addon. 

Any changes? 
https://github.com/me-no-dev/EspExceptionDecoder

regards
   Dominik

jantje, 2016-10-07:
If there is enough interest from sponsors I could take a look.

ghislainborremans, 2017-04-17:
i agree, would be nice addon

jantje, 2017-04-17:
We have a sponsor ;-)

jantje, 2017-04-25:
If you get an exception you want to dump in this Exception decoder can you paste it here.
I still do not have a clue what this is all about.
Or a way to get exceptions would be good to.

ghislainborremans, 2017-04-25:
the command to execute ( batch file in windows) 
java -jar D:\Electronica\EspStackTraceDecoder.jar D:\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe D:\Electronica\S_Workspace\Intiles\Release\Intiles.elf D:\Electronica\dump\dump.txt

the dump.txt: i copied this from the serial console output

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffcc0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe60:  00000000 3fffbf8c 0000002c 40100705  
3ffffe70:  00000000 3fffbf8c 4024b4f0 4010754c  
3ffffe80:  00000008 01a493d1 60000600 40216af8  
3ffffe90:  0000004c 402191ab 00000018 0000000d  
3ffffea0:  00000001 00001388 3ff20200 ffffffff  
3ffffeb0:  3ffffed0 4021ce53 3ffee6bc 00000008  
3ffffec0:  40242b0a 00000008 3ffeee68 00000064  
3ffffed0:  4023e8b1 3ffeeddc 3ffe83f8 3ffe83f8  
3ffffee0:  3fff2597 3fffdad0 3fff1dc4 0000000d  
3ffffef0:  4021e9ef 3ffeed90 3ffeee68 01a493d1   
3fffff00:  00000001 4023fb1c 3ffee740 40211c74  
3fffff10:  4021e6a4 4021e678 40220468 00000000  
3fffff20:  4022043d 4021e678 3fff09fc 40206fcd  
3fffff30:  3ffee794 40220404 3ffeeda8 3ffefb90  
3fffff40:  4021e66e 3ffee6bc 3fff0848 40211d00  
3fffff50:  4021e615 3ffeeda8 3ffefb90 01a493d1  
3fffff60:  4021eac8 402204b2 4021e6b4 60000600  
3fffff70:  40231405 3ffefb90 3fff2a04 40231405  
3fffff80:  4023144a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffefba0 3fffdad0 3fff0ab8 4021002b  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

jantje, 2018-06-02:
@william-ferguson-au 
Nice to hear you want to sponsor. I don't know what you mean with sloeber workspace. There is a eclipse plugin which used to be called "the arduino eclipse plugin" which was renamed to Sloeber for legal reasons.
The Sloeber plugin needs to be added to a existing eclipse install. 
As many users had a problem getting all required plugins; we made a Sloeber product which is a eclipse install (cdt + some more stuff) bundled with Sloeber plugin.
The core functionality is in the Sloeber plugin so from that point of view the product and the plugin are exactly the same. From a deployment (1 download) and merchandising (different splash screen) point of view they are different.

hreintke, 2018-06-02:
Hi,

I configured a external tool to have this "feature".

[image]

The dump.txt file I have in the project directory. In that I copy/paste the stacktrace.

Should be possible to have it all project related using the variables but did not get that to work.

william-ferguson-au, 2018-06-02:
Well, I ditched Eclipse for greener pastures 10 years ago. But I think we're talking about the same thing. I guess I'm using the Sloeber product.

[image]

As @hreintke pointed out the EspStackTraceDecoder already provides the processing, it's just a matter of incorporating that into the Sloeber UI. 

Minimum would be a window to paste the stacktrace, invoke the decoder and show the result.
Better would be to annotate the Serial Monitor view with an actionable link/button whenever a backtrace is written than open a window showing the decoded trace.

And thanks for the work on Sloeber. The Arduino IDE was driving me nuts, it's not a SW IDE IMHO.

You need to point me at your patreon. I couldn't find a link anywhere.

william-ferguson-au, 2018-06-03:
Thanks @TrippyLighting - supported

jantje, 2018-06-03:
@william-ferguson-au 
Thanks for becoming a patron :-)

@all
I have never used this feature so I still have a bit of a problem getting my head around this.
Lets start from this proposition
>Better would be to annotate the Serial Monitor view with an actionable link/button whenever a backtrace is written than open a window showing the decoded trace.

There are still questions for me given the above description
1) Do I save the full (?) content of the serial monitor and/or the selected content?
2) Can I save the serial content in a file in the temp folder? (this will be best for version control)
3) What if there are multiple serial connections. Can we call this non supported?

william-ferguson-au, 2018-06-03:
No need to save the full content of the serial monitor. You are only interested in output like:

`Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (pthread) 
Register dump:
PC      : 0x40144c65  PS      : 0x00060836  A0      : 0x80144638  A1      : 0x3fff1880  
A2      : 0x00000000  A3      : 0x0000001f  A4      : 0x00000001  A5      : 0x3f401d35  
A6      : 0x3fff1c10  A7      : 0x00000008  A8      : 0x3f401d16  A9      : 0x3fff1a74  
A10     : 0x3fff1890  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x40144c65:0x3fff1880 0x40144635:0x3fff1b90 0x400d3194:0x3fff1c50 0x400d48f8:0x3fff1e80 0x400d4a95:0x3fff2080 0x400d55e5:0x3fff20c0 0x400d39a6:0x3fff20e0 0x400d3b65:0x3fff2120 0x401b6494:0x3fff2180 0x4017d4b5:0x3fff21a0 0x4017aff3:0x3fff21d0
`

And of most importance to the EspStacktraceDecoder is this line:
`Backtrace: 0x40144c65:0x3fff1880 0x40144635:0x3fff1b90 0x400d3194:0x3fff1c50 0x400d48f8:0x3fff1e80 0x400d4a95:0x3fff2080 0x400d55e5:0x3fff20c0 0x400d39a6:0x3fff20e0 0x400d3b65:0x3fff2120 0x401b6494:0x3fff2180 0x4017d4b5:0x3fff21a0 0x4017aff3:0x3fff21d0
`

I don't understand point 2.

Multiple serial connections is not required.

If possible, I would suggest turning the Backtrace line into a link that opens a window with the stacktrace decoded.

jantje, 2018-06-03:
Point 2 is about where the dump file will be stored. If it is in the project it will be picked up by version control tools. Having it in the temp folder is a safer way, I guess

>If possible, I would suggest turning the Backtrace line into a link that opens a window with the stacktrace decoded.

I can look into this but I have no clue how to do such a thing.

hreintke, 2018-06-03:
A quick first remark : 
Beware, the output above is for ESP32, for ESP8266 the format is different.
Not sure whether the EspStackTraceDecode supports boths boards.

There is EspExceptionDecoder which is started as a Arduino IDE tool, that is window based and supports both.

jantje, 2018-06-03:
@hreintke 
Thanks for the heads up.
I think the most generic will be to select text in the serial monitor and press a button that saves the selected text in a dedicated location (FI ${project_loc}/${config_name}/serialdump.txt to avoid version impact) and runs a "external tool"
With a generic format based on what you provided that should be very generic and easy to adjust to other tools
`-jar c:\temp\EspStackTraceDecoder.jar ${eclipse_home}\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe ${project_loc}/${config_name}/${project_name}.elf ${project_loc}/${config_name}/serialdump.txt`

Not sure I can make this work but I'm trying

jantje, 2018-06-04:
Selecting t(ext in the serial monitor and right clicking with a project selected in the project explorer will create a serialdump.txt file in the config folder(normally that is release) of the selected project.
Together with the external tool that should make life a lot easier already.
I'm looking for a way to popup the external commands on the rightclick.

william-ferguson-au, 2018-06-04:
Sounds good!

jantje, 2018-06-05:
Even better part one is already available in the nightly ;-)

hreintke, 2018-06-06:
Hi Jantje,
How is it supposed to work.

My test environment/actions are.

I have esp8266 attached 
- link the serial monitor to the correct comport
- The esp8266 prints stacktrace.
- Disconnect esp8266 from comport to prevent continuous dumps.
- Selecting the stacktrace test.

With right click : 
- When no project selected : 
`There are no projects selected in the project explorer or all selected projects are not arduino projects. Are you in the Arduino perspective?`
- When project selected -> nothing.

jantje, 2018-06-06:
The file ${project_loc}/${config_name}/serialdump.txt is created/updated with the content of the selection in serial monitor.
However if eclipse doesn't know the file yet you need to refresh (F5) the project for eclipse to find it.

hreintke, 2018-06-07:
Retested.

1/ I do get the serialdump.txt in the release directory (after F5)
2/ When running with `-jar c:\temp\EspStackTraceDecoder.jar C:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe ${project_loc}/release/${project_name}.elf  ${project_loc}/release/serialdump.txt` it runs, with `-jar c:\temp\EspStackTraceDecoder.jar C:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe ${project_loc}/${config_name}/${project_name}.elf  ${project_loc}/release/serialdump.txt` I get `Project name should be specified as variable argument for ${config_name} variable`
3/ Can't get the output file. Neither by adding it to the -jar... arguments (which seem logical as it is no option   in the EspStackTraceDecoder) nor by adding it as a outputfile in the common tab of external tool configuration.

jantje, 2018-06-07:
My bad. Apparently you can not use  ${config_name} for a replacement for release
Can you try this command?
 
-jar c:\temp\EspStackTraceDecoder.jar C:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe ${project_loc}/release/${project_name}.elf ${project_loc}/release/serialdump.txt it runs, with -jar c:\temp\EspStackTraceDecoder.jar C:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\xtensa-lx106-elf-addr2line.exe ${project_loc}/release/${project_name}.elf ${project_loc}/release/serialdump.txt

hreintke, 2018-06-07:
Unless I am overlooking something, both commands in your last comment are identical to the one I mentioned that worked.

jantje, 2018-06-07:
They are not
bad
${project_loc}/**${config_name**}/${project_name}.elf ${project_loc}/release/serialdump.tx
good?
${project_loc}/**release**/${project_name}.elf ${project_loc}/release/serialdump.txt

hreintke, 2018-06-07:
From my first comment : 

> ... ${project_loc}/release/${project_name}.elf ${project_loc}/release/serialdump.txt it runs

jantje, 2018-06-07:
great :-)
I just hoped we could use ${config_name} instead of release so it would work fro people using multiple configs but apparently not just like that.
Getting a list of external tools popping up seems to be,... not so easy as you can see here https://stackoverflow.com/questions/50689269/how-to-get-the-list-of-external-tools-in-eclipse?noredirect=1#comment88438720_50689269

jantje, 2018-06-19:
I got the list of commands,... it seems creating a pop-up menu in eclipse is not yet in my set of skills :-(

erdemuncuoglu, 2019-08-29:
>I just hoped we could use ${config_name} instead of release so it would work fro people using multiple configs but apparently not just like that.

In fact we can. I'm using it like this:

${project_loc}/**${config_name:${project_name}}**/${project_name}.elf ${project_loc}/**${config_name:${project_name}}**/serialdump.txt

jantje, 2019-08-29:
thanks for the input.

wimjongman, 2019-08-30:
Why not use a java tempfile? 

https://docs.oracle.com/javase/7/docs/api/java/io/File.html#createTempFile%28java.lang.String,%20java.lang.String%29

jantje, 2019-08-30:
@wimjongman 
I'm not sure what the java tempfile will add.

If I remember correctly this is a request to run an external tool on part of the content of the arduino monitor output. 
I succeeded in saving the data in a file which is located in the "active configuration folder" with a fixed name by selecting text in the serial monitor and right clicking with a project selected in the project explore
I wanted a popup to appear at the right click to select the tool that needs to be run but I did not succeed to.
Therefore the workaround is to run a eclipse command. 
Whether the command is run directly or indirectly, the command needs to know the configuration to locate the file.
This can be hard coded as release (which will work in 99% of the cases) but according to @erdemuncuoglu ${config_name:${project_name}} should cover 100% of the cases.

wimjongman, 2019-08-30:
Can you provide a github link to the code?

jantje, 2019-08-30:
Here is the code saving the monitor content
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.ui/src/io/sloeber/ui/monitor/views/SerialMonitor.java#L317
I tried to start the launcher in branch start_launcher_from_serial_monitor#542 but I never got anywhere