Sloeber/arduino-eclipse-plugin #543: V3.1 does not install standard Arduino tools (i.e. GCC)

edogaldo, 2016-10-07:


edogaldo, 2016-10-07:
Note: switching the platform version (i.e. from 1.6.14 to 1.6.11 and then back to 1.6.14) installed the missing tools.

jantje, 2016-10-07:
Do you have logs that may show what went wrong?
Didn't you get an error your path is to deep? (windows still has a maximum path depth)

edogaldo, 2016-10-07:
Where could I find logs?
The fact that switching the version solved the problem tells me it shouldn't be due to the path length..

jantje, 2016-10-07:
>Where could I find logs?
The fact that switching the version solved the problem tells me it shouldn't be due to the path length..

It does not tell me this.

The process you describe (what you call switching) are actually 4 completely separate actions.
Deinstall 1.6.14 install 1.6.11 deinstall 1.6.12 and install 1.6.14. I'm not even sure it will happen in this order. 

Have you tried to add hardware to the plugin by copying files in [Eclipse install]/ArduinoPlugin?
If so it is normal Arduino wil not be installed as these are only installed if no boards are available.

edogaldo, 2016-10-07:
This has nothing to do with issue #545
I was performing first installation of the v3 plugin on a fresh eclipse (neon) installation.
After installing the plugin the arduino subfolder (the one containing the avr-gcc) was missing and I could not compile.
Even uninstalling and reinstalling the plugin added the missing subfolder..

edogaldo, 2016-10-07:
Even uninstalling and reinstalling the plugin did not add the missing subfolder..

jantje, 2016-10-07:
>Even uninstalling and reinstalling the plugin did not add the missing subfolder..

This will do nothing indeed.

jantje, 2016-10-29:
Did you install from the nightly?
If so this is probably a duplicate of #551

jantje, 2016-11-01:
I have seen that the error handling in this part of the code is not quite good. I fixed this in V4 #570.
I assume something went wrong on your system and it did not get reported.
So I propose to close this one.

closed 2016-11-01 by jantje