Sloeber/arduino-eclipse-plugin #544: .ino.cpp tries to include all files

nicoverduin, 2016-10-07:
I have this test sketch for Lora (further not relevant btw) with the following code:
`#if defined(__AVR__)
#include <avr/pgmspace.h>
#include <Arduino.h>
#elif defined(ARDUINO_ARCH_ESP8266)
#include <ESP.h>
#elif defined(__MKL26Z64__)
#include <Arduino.h>
#else
#error Unknown architecture in aes.cpp
#endif

#include "lmic.h"
#include "hal/hal.h"
#include <SPI.h>`

However the .ino.cpp includes all the files
`//This is a automatic generated file
//Please do not modify this file
//If you touch this file your change will be overwritten during the next build
//This file has been generated on 2016-10-07 15:53:00

#include "Arduino.h"
#include <avr/pgmspace.h>
#include <Arduino.h>
#include <ESP.h>
#include <Arduino.h>
#include "lmic.h"
#include "hal/hal.h"
#include <SPI.h>
void os_getArtEui (u1_t* buf) ;
void os_getDevEui (u1_t* buf) ;
void os_getDevKey (u1_t* buf) ;
void onEvent (ev_t ev) ;
void do_send(osjob_t* j);
void setup() ;
void loop() ;

#include "LoraLMIC151Test.ino"`

ESP.h should not be included however. Now I just manually comment this file in the .ino.cpp.

jantje, 2016-10-07:
This is a known limitation. #478 
Put the ifdef include structure in a header file and the problem is solved.

jantje, 2016-10-07:
Hi nico. Why do you want to use .ino files?

closed 2016-10-07 by jantje

nicoverduin, 2016-10-07:
No particular reason but makes it easier for the NoNo's to use my programs :) I just came across this situatie and wanted to let you know. So don't bother to fix it. I can manager :)