Sloeber/arduino-eclipse-plugin #545: plugin v3: How to install custom cores?

edogaldo, 2016-10-07:
I'd like to add the stm32duino core (https://github.com/rogerclarkmelbourne/Arduino_STM32) for STM32 MCUs. This core at the moment has not a json file.
I tried to put the core in the "private hardware path", this way the plugin can somewhat see it but variable **{runtime.tools.arm-none-eabi-gcc.path}** seems wrong (empty) leading to errors: 
- "Program "\bin\arm-none-eabi-g++" not found in PATH"
- "Program "\bin\arm-none-eabi-gcc" not found in PATH"

Notes: 
- in standard Arduino installations the stm32duino core uses the toolchain of the Arduino Due core (just need to install the Arduino Due core to be able to successfully compile with stm32duino)
- I installed the Arduino Due core which drove the installation of the arm-none-eabi-gcc toolchain 
- I can successfull compile with the Arduino Due core

jantje, 2016-10-07:
I think this is a duplicate. Can you upgrade to the nghtly?
Use the update process. Do not use the download nightly product as that will install V4 which is still pretty unstable.

jantje, 2016-10-07:
#516

edogaldo, 2016-10-07:
Thanks @jantje, yes, the problem seems similar.
How to upgrade to the nightly?

edogaldo, 2016-10-07:
Current plugin version currently installed is 3.1.0.201608012005
If I do 

Help > Installation Details > select "Arduino Eclipse Extensions" > click Update

I get message "No updates were found in available software sites"

edogaldo, 2016-10-07:
Ok, found the way to install the latest nightly build: 3.1.0.201609060208

I can confirm this fixes the problem! 

Thank you very much and best regards, E.

closed 2016-10-07 by edogaldo

jantje, 2016-10-07:
>Help > Installation Details > select "Arduino Eclipse Extensions" > click Update
I get message "No updates were found in available software sites"

That is because you do not have http://eclipse.baeyens.it/nightly active in your update sites.