Sloeber/arduino-eclipse-plugin #546: platform.txt of readbear duo is not parsed properly, making the build fail.

jantje, 2016-10-08:
Basically I do not see how the archiving activities are linked to the linking/combining activities.

Some more detail is here http://discuss.redbear.cc/t/redbear-duo-package-redbear-index-json-should-list-boards-as-an-array/1024.

jantje, 2016-10-19:
I found a bug in Sloeber in relation to this issue.
Nasically the archive reference is removed from the combine recipe but the archive is not added anymore. 
The above fix add the archive back to the command.

Note that redbear has to modify the platform.txt for this issue to be fixed.

jantje, 2016-10-19:
This change make that it works in both Arduino IDE and sloeber.

```
## Create archives
archive_file=libcore.a
archive_file_path={build.path}/{archive_file}
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.device.define} -g3 -gdwarf-2 -Os -mcpu={build.mcu} -mthumb -fno-builtin {compiler.platform.define} {compiler.system_version.define} -Werror {build.variant_base_include} {build.variant_extra_include} -I. -MMD -MP -MF "{build.path}/{build.project_name}.elf.d" -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 {compiler.baudrate.define} -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc {compiler.image.define} {compiler.module.define} {object_files} --output "{build.path}/{build.project_name}.elf" -Wl,--whole-archive "{runtime.platform.path}/cores/RedBear_Duo/redbear_sdk/X_lib/STM32F2xx_Peripheral_Libraries.a" -Wl,--no-whole-archive -nostartfiles -Xlinker --gc-sections {build.ld_include2} {build.ld_include3} -L{build.path} "-L{runtime.platform.path}/cores/RedBear_Duo/redbear_sdk/X_lib" -Wl,--whole-archive "{archive_file_path}" -lhal-dynalib -lservices-dynalib -lsystem-dynalib -lrt-dynalib -lwiring -lcommunication-dynalib -lplatform -lwiring_globals -Wl,--no-whole-archive -lnosys {build.ld_include1} -L. "-T{build.variant.path}/{build.ldscript}" -Wl,--defsym,USER_FIRMWARE_IMAGE_SIZE=0x40000 -Wl,--defsym,USER_FIRMWARE_IMAGE_LOCATION=0x80C0000 -Wl,-Map,{build.path}/{build.project_name}.map -lstdc++_nano -lm -Wl,--start-group -lgcc -lg_nano -lc_nano -Wl,--end-group -Wl,--start-group -lgcc -lc_nano -Wl,--end-group
```
Lets hope redbear changes the platform.txt

closed 2017-01-05 by jantje

frmdstryr, 2017-03-11:
Was this fixed? Can you create a diff / patch for this file to fix it?

jantje, 2017-03-11:
As far as I know redbear fixed this issue