Sloeber/arduino-eclipse-plugin #549: CPU usage is too high

gundamlh, 2016-10-10:
Dear community,

The CPU usage of Sloeber IDE (stable V3.1) on Ubuntu 14.04 LTS (64 bits) is incredibly high: [screenshot of htop](http://pasteboard.co/dh7e4eX8f.png).

I've tried following nightly builds: 
1. 2016-10-04	linux64.2016-10-04_04-17-04.tar.gz
2. 2016-09-22	linux64.2016-09-22_04-17-11.tar.gz

Both have the same issue. What is happening? I did not install any other plug-ins.

Thank you and Best regards,
Honig

jantje, 2016-10-10:
I notice a long (seconds) high cpu usage at every startup of sloeber. But then things go back to normal.
I have noticed this since I use neon. I have seen reports of other people (not using sloeber but using CDT) but I never got a clear answer nor a easy to reproduce way. 
In other words I do not think this is Sloeber (as in the plugin) related but caused by one of the components Slober is build upon.

gundamlh, 2016-10-11:
Thank you for the quick reply! Forget it, the issue disappeared after Ubuntu update this morning. It is quite funny...

jantje, 2016-10-11:
CPU issues are hard to track back to a root cause.
Thanks for the feedback that your issue it is fixed and not (really) Sloeber related.
And don't forget : You need(ed) our support but we also need your support https://www.patreon.com/jantje

closed 2016-10-11 by jantje