Sloeber/arduino-eclipse-plugin #551: Can't install Arduino Board Platforms

DerEisvogel, 2016-10-16:
Hello,

I am trying to install the ESP8266 Platform but sadly it always fails with an error:
'Adopting Arduino Board Platforms' has encountered a problem.

> Installing Arduino Board Platforms
> OK
> OK
> OK
> OK
> downloading make.exe
> Invalid username/password

I have tried adding other boards and they all fail with the same error message.

One thing to mention: I am working on Windows and when I start Eclipse it gives me the error "bash and sh found on you system. The plugin will not work properly."
I guess the error was right. :-D

Anyone has an idea how to fix this?

Regards,
Eisvogel.

jantje, 2016-10-16:
You do have to fix the bash and sh error.
Which version of the plugin are you using?

jantje, 2016-10-16:
I see there is regression due to #393 which makes the makefile download fail.
Sloeber only tries to download the file if it does not yet exists.
So a obvious workaround is to copy the file yourself from ftp://ftp.equation.com/make/32/make.exe to the [eclipseinstall]/arduinoPlugin/tools/make folder

jantje, 2016-10-29:
Closing as this is a regression

closed 2016-10-29 by jantje