Sloeber/arduino-eclipse-plugin #554: Sloeber Programming issue

NielsP90, 2016-10-25:
Hello,

Yesterday I downloaded Sloeber (x64 from http://eclipse.baeyens.it/stable.php?OS=Windows )
Everything is running fine. As far as I can see, it builds code fine. But...it doesn't seem to update on the board. I don't see the flashing LEDs on the Arduino board like I did with the Arduino IDE. 
I am using Arduino Uno.
Did anyone experience the same issue? What am I doing wrong. 

With kind regards,

Niels Puts

[Build log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/549985/Build.log.txt)
[Upload log.txt](https://github.com/jantje/arduino-eclipse-plugin/files/549986/Upload.log.txt)

jantje, 2016-10-25:
Investigating this issue I saw Arduino released toolchain version 1.14 which uses a new avrdude tool "6.3.0-arduino6/bin/avrdude " of which I assumed it has the same problem as toolchain 1.12 see issue #510 
But after testing I did not have that problem.
It works fine on my system. I don't know why but at your computer the uno does not get reset. 
My upload log starts with
```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use COM3
Ending reset
...
```
You are missing the reset part in your log.
We had some issues with this reset so it may have been fixed in the nightly. 
Can you update the plugin (do not download the nightly as then you will go to V4 which is not stable enough yet) and try again?

jantje, 2016-10-29:
@NielsP90 Please provide the requested input.

NielsP90, 2016-11-01:
Sorry for the delayed response. My time was consumed by other projects. 
I tried deleting the Sloeber directory and "reinstalling" with a freshly downloaded image. That didn't work. What did work however was setting the AVR toolchain from 1.6.14 to 1.6.13. Now it works just fine. 
Thanks for your help!

jantje, 2016-11-01:
Thanks for the feedback.
I'm closing this as a one of.

closed 2016-11-01 by jantje