Sloeber/arduino-eclipse-plugin #555: Compile issues when switching between (some) boards in the same project

sergiuburian, 2016-10-25:
First things first: Thanks a lot for the good work you have done so far. It really makes our life easier when developing for embedded, especially for the guys like me coming to embedded from the traditional software development and tools. It takes the fear away ;)

Now to the issue: Kindly ask for the following: when switching between the boards in the same Eclipse project, please delete the previous linked resources (folders) under /core and create new ones that match with the newly selected board. I am encountering strange compile issues because when switching between the boards because the linked resources under "core" are not refreshed. E.g. when switching from Adafruit Feather M0 to Teensy 3.2, the linked folder to "variant" is kept, pointing to the M0 folder. I'm sure you guess what happens :) Basic types not found, millis not found, because the variant does not compile and match with core.

Motivation: when starting a new project with some other goal that playing, you often do not know which board you should take, so you try more than one. Also, when planning to share the project source, you want this to work on several board types. Both these cases require the development on a single software line, so a single project (for convenience), ready for multiple platforms. 
Now I understand that multi-platform  build processes, build targets and /release folders means some serious work, therefore, I am asking for facilitating switching between different boards in the same Eclipse project in a simple manner.

Thank you,

Sergiu

jantje, 2016-10-25:
The functionality you are referring to is implemented. You can even have multiple configurations each with their own board. And indeed it wasn't simple to do (thanks for the compliment) :-)
However there seems to be a bug that when you swap between 2 boards -where one has a variant and the other not- the old variant remains.
Simply deleting the core/variant folder (when moving to a board not needing a variant) fixes the problem

sergiuburian, 2016-10-25:
Hi,

Thanks for the prompt answer. I should have also mentioned the workaround, will do better next time :)

Cheers,

Sergiu

closed 2017-01-17 by jantje