Sloeber/arduino-eclipse-plugin #556: Arduino Sketch Project Build fails with "recipe for target 'arduino.ar' failed"

rthomas67, 2016-10-26:
# Steps to reproduce on Windows 7 - 64bit.

Installed by unzipping from sloeber 3.1 tar.gz download for Windows x64
Java JRE Version is 1.8.0_45
Started by running sloeber-ide.exe, initializing a workspace, etc. (normal eclipse stuff).

MENU: File->New->New Arduino Sketch
Name: my_sketch_name
Opened my_sketch_name.ino to see standard empty setup() and loop()
MENU: Project->Clean...
    SELECT: Clean all projects
    SELECT: Start a build immediately
    SELECT: Build the entire workspace
    CLICK: OK

Build fails.

# Error Information
Eclipse "Problems" pane has 2 errors
1) Description: make: *** [arduino.ar] Error1,  Resource:my_sketch_name, Type: C/C++ Problem
2) Description: "recipe for target 'arduino.ar' failed,  Resource:makefile, Path: /my_sketch_name/Release, Location: line 73, Type: C/C++ Problem

generated makefile has an error on line 73 which is:
    @echo 'Starting archiver'

If it 's relevant, the Eclipse error log has several messages like this:
!ENTRY it.baeyens.arduino.core 1 0 2016-10-25 19:36:57.567
!MESSAGE The folder C:\Users\myuserid\Arduino\hardware' does not contain any files.

jantje, 2016-10-26:
Can you provide the info of this error log message
!ENTRY it.baeyens.arduino.core 1 0 2016-10-25 19:36:57.567

Can you provide the cdt build console output?

FYI
>!MESSAGE The folder C:\Users\myuserid\Arduino\hardware' does not contain any files.

This means Sloeber did not find any hardware in this folder.

rthomas67, 2016-10-27:
Didn't notice this before, but upon taking another look at the project and the build console I observed that the first part of the path as reported in the console error message was D:/rtfile_local/...   However, the actual filename (using Windows slashes) starts with  D:\$rrtfile_local\...

It appears that filenames/paths containing a "$" are not properly quoted or escaped in the generated makefile and it seems "$r" is being replaced by "".  Guessing that's because there is no $r variable defined.

I confirmed that the "$" character was the proverbial spanner in the works by moving the workspace/project_folder out of the directory that contained the "$" and the build worked.

Here is the command, as generated into the makefile.

"C:\javaapp\eclipse_arduino_sloeber_3.1\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-gcc-ar" rcs "D:/$rrtfile_local/eclipse_sloeber_ws/test_sketch/Release/arduino.ar"  arduino.ar $(AR_OBJ)

Here is the line from the build console where that failed.

"C:\javaapp\eclipse_arduino_sloeber_3.1\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-gcc-ar" rcs "D:/rtfile_local/eclipse_sloeber_ws/test_sketch/Release/arduino.ar"  arduino.ar  .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o 
c:\javaapp\eclipse_arduino_sloeber_3.1\arduinoplugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: D:/rtfile_local/eclipse_sloeber_ws/test_sketch/Release/arduino.ar: No such file or directory
makefile:73: recipe for target 'arduino.ar' failed
make: *** [arduino.ar] Error 1

Now if I could just figure out why this toolchain is SOOOO miserably slow on Windows.
"20:49:05 Build Finished (took 1m:46s.65ms)"

Same exact "empty" project finished a build in about 3 seconds on a friend's MacBook Pro.  Same Windows machine handles nearly everything else as fast or faster than the Mac.

jantje, 2016-10-27:
I'm not surprised $ signs in the path cause problems. The whole boards.txt and platform.txt are converted to environment variables that use $ to identify environment variables. I'm using the format ${environment variable name} but I can not instruct the environment variable tool that I use to only process this type of variable.
Honestly I never considered $ in the pathname.

As to your compile speed. I think there is a local issue on that system. Compile speed is normally lots faster then the Arduino IDE.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2016-10-29:
closing this as it is tagged as known limitation

closed 2016-10-29 by jantje