Sloeber/arduino-eclipse-plugin #557: rename scope to plotter.

jantje, 2016-10-26:
The scope is actually a plotter.
Given that we have incompatibilities due to the refactoring I renamed all scope (when it stands for oscilloscoop) to plotter.

universpharmacie, 2016-12-30:
Just a "peripheral" question concerning the oscilloscope/plotter: Actually, it (unfortunately) behaves more like a plotter than like an oscilloscope. Neither the y axis (amplitude) nor the x-axis (time) can be scaled. There is no triggering functionality. That's very sad, especially when one is developing real time audio applications or checking serial communication, where the plotter is pretty useless compared to a "true" oscilloscope/logic analyzer. Are there plans to improve that one day?

jantje, 2016-12-30:
@universpharmacie 
Thanks for the input
You can scale/drag the y-axis but it is with right/left click and drag and as far as I know that doesn't work on mac. 
When @mjmeijer levelled up and feels confident I'm sure he will be happy to look at an alternative mac approach.
I think logic analyser will never be on my todo list. however it is eclipse so I checked and there seems to be a project called impulse. If you check it out and think it is useful I could make a feeder.
Setting a trigger level can't be that hard to implement but it is not on my todo list right now and maybe it is no longer needed if impulse is "connected".

PS: Did you know you can save the data in csv format by double clicking?

universpharmacie, 2016-12-30:
Here is what I'd like to see ideally in the scope window... 
[image]

jantje, 2016-12-31:
This is a full scope functionality which is different from the potter functionality.
As I'm a software guy I don't think I'm the right person to develop a scope but there are plugins that mention and look like a scope and a logic analyser on the eclipse marketplace.
I would really appreciate you have a look at it to see whether you like it. As far as I can see Sloeber must be able to feed them with data.
You can find one here
https://marketplace.eclipse.org/content/impulse

universpharmacie, 2016-12-31:
I knew your answer before, but thank you. I just thought that since we are still in the holiday period, I could add to my wishlist. ;-) 
I will for sure look into that impulse plugin in the next days. Happy new year, Jantje!

jantje, 2016-12-31:
txs.
Happy new year to you :-)
I do think you will have better service using a existing scope implementation then if I need to build something. After all this is the eclipse way of thinking. linking tools together so everybody can focus on his/her part and the end user gets a integrated full   featured ide

closed 2017-01-17 by jantje