Sloeber/arduino-eclipse-plugin #565: Use the nebula package code directly instead of the copy.

jantje, 2016-10-30:
In #557 the cope is now plotter but also the plotter class has been added to nebula.
Now sloeber uses a copy of the plotter class embedded in Sloeber but the correct way is to use nebula directly so that changes/improvements get incorporated in both nebula as Sloeber.
To do so the copy of plotter needs to be removed from Sloeber and a link to the nebula code must be added.
This can only be done after the change adding the plotter class to nebula has been marked as ready.

wimjongman, 2016-10-30:
> @wimjongman <https://github.com/wimjongman>.

In master?

jantje, 2016-10-30:
I wouldn't know where else

wimjongman, 2016-10-30:
I thought we were using gitflow but never mind. I miss some .project files. for example in io.sloeber.ui which prevens me from checking out the projects.

io.sloeber.ui
io.sloeber.ui.nl1
io.sloeber.monitor

jantje, 2016-10-30:
I'm having a hard time to find a way to check them in

wimjongman, 2016-10-30:
I will take care of it.

jantje, 2016-10-30:
I found it. apparently the main gitignore contained .*

jantje, 2016-10-30:
I pulled in the changes and it doesn' compile anymore in my eclipse :-(
Seems he can't find the nebula plotter.

wimjongman, 2016-10-31:
You have to install the nebula oscilloscope.

jantje, 2016-10-31:
I'll try that and add that to the readme once I know how :-)

wimjongman, 2016-10-31:
This only applies when the plugins installed separately. The IDE will
include the Scope from Nebula automagically.

For plug-in installers, the nightly repository of Nebula is here [1] which
is documented here [2]:

[1] http://download.eclipse.org/nebula/snapshot
[2] https://www.eclipse.org/nebula/downloads.php

jantje, 2016-10-31:
The update to the readme I did was for the installation of the development environment for sloeber.
As io.sloeber.ui states it is dependent on org.eclipse.nebula.widgets.oscilloscope.multichannel I hope this one will be installed with the plugin.
I'll test whether this assumption is true :-)

jantje, 2016-11-02:
I tried to install sloeber in a new cdt install and got the following error message:
```
Cannot complete the install because one or more required items could not be found.
  Software being installed: Sloeber 4.0.0.201611010210 (io.sloeber.feature.feature.group 4.0.0.201611010210)
  Missing requirement: Ui 4.0.0.201611010210 (io.sloeber.ui 4.0.0.201611010210) requires 'package org.eclipse.nebula.widgets.oscilloscope.multichannel 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Sloeber 4.0.0.201611010210 (io.sloeber.feature.feature.group 4.0.0.201611010210)
    To: io.sloeber.ui [4.0.0.201611010210]
```
So I guess we only need to point it to the downloadsite from nebula to get this to work

wimjongman, 2016-11-02:
We can do three things.

1. Leave it like it is and then you have to add the nebula repo like I explained earlier
2. Include the jar inside sloeber (we call this a fat plugin), against the grain of OSGi
3. Include the scope in the plugins. 

I guess the last option is the most convenient. I will see if I can make that happen.

jantje, 2016-11-02:
Will test tomorrow :-)

jantje, 2016-11-05:
After very long trying (hours) it failed but this time no error message.
I'll try locally here to get it to work.

jantje, 2016-11-05:
The long wait was caused by a bad url.

jantje, 2016-11-08:
As I don't see how to do this I asked the question on stackexchange.
http://stackoverflow.com/questions/40444369/eclipse-plugin-development-plugin-requires-second-update-site-for-installation
Not much response :-(

wimjongman, 2016-11-10:
Sorry I missed this. I thought I solved it by adding the nebula repo to the extra update site [1]. Doesn't it do it? If not I will include it in the plugins.

[1] https://github.com/wimjongman/arduino-eclipse-plugin/commit/cab47826c8584dac1e489aea9cece805be0fd811

jantje, 2016-12-04:
I just checked it. Still fails :-( and no answer on my question on stackoverflow.
I tried to give a bounty but I have not enough credit I guess.

wimjongman, 2016-12-05:
I'll fix it tomorrow first thing.⁣

Sent from TypeApp ​

jantje, 2016-12-09:
Tested and it worked :-)

closed 2017-01-17 by jantje