Sloeber/arduino-eclipse-plugin #569: Sloeber symbols OUTPUT, HIGH and LOW cannot be resolved.

stevejoung, 2016-11-01:
Hi, I have installed the Sloeber bundle from here http://eclipse.baeyens.it/stable.php?OS=Linux onto my Ubuntu 16.04 laptop. I can open the IDE and start a new sketch and then select a sample sketch such as Blink. However when I look at the Blink.ino it shows syntax errors against the code. The "Problems" tab at the bottom of the IDE also shows the errors, which are that the symbols OUTPUT, HIGH and LOW cannot be resolved. 

I can uplaod the sketch to the Arduino anyway, but how can I remove these error messages and warnings?

stevejoung, 2016-11-01:
OK. If I include the Arduino library (#include <Arduino.h>) by hand in the Blink.ino file, it works. But I thought libraries were automatically included in Sloeber?

jantje, 2016-11-01:
Arduino.h is not a library but a core.
Read this if you want to understand it at full length.
https://www.patreon.com/posts/there-are-bugs-5507246

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2016-11-01 by jantje