Sloeber/arduino-eclipse-plugin #572: Cannot create new project twice in nightly build

TukaP, 2016-11-02:
Creating second new project no options on second and three window are available. What you need to do is reselect Board.txt file all options are back then.

jantje, 2016-11-02:
I havn't seen this yet in the new project wizard. Does a resize help?
Note the nightly=V4 is still pretty bleeding edge.

jantje, 2016-11-12:
@TukaP 
Is this fixed in the latest nightly? If not can you give some steps to reproduce?

TukaP, 2016-11-13:
Hi Jantje,

IHMO the issue persists. When I want to create second arduino project
then I get Sketch template folder empty (see pist. bellow). What I need
to do is gat back to first step ans reselect Board.txt file and set up
pramaters next to. Everything is OK then. Templates are visible etc.

I am not sure what build I use but refreshed right now from Eclipse
Marketplace.

Best regards

Paul

*

Pavel Tuka*
Dne 12.11.2016 v 23:57 jantje napsal(a):

jantje, 2016-11-13:
This is becoming very confusing to me.
The original title said: "... in nightly build" So I assumed it was V4.
However now you state" but refreshed right now from Eclipse Marketplace." The market place does not link to the nightly.
I assume "see pist. bellow" should have said "see pictures bellow". But I see no pictures below.
So what is your OS, what is your installed eclipse version what is your installed version of sloeber and please explain step by step how to reproduce the problem.

jantje, 2016-12-04:
@TukaP closing as to long open without input

closed 2016-12-04 by jantje