Sloeber/arduino-eclipse-plugin #574: Add support for Oxygen

wimjongman, 2016-11-02:


jantje, 2016-11-12:
Should we upgrade the product to oxygen?

jantje, 2016-12-04:
@wimjongman Do we keep this open to move the product to oxygen or can we close this as it doesn't impact the end user??

wimjongman, 2016-12-05:
This is the ticket to prepare for Oxygen. We will switch once it is out in
2017.

mattb5906, 2017-01-02:
Just being nosey: what is Oxygen?

jantje, 2017-01-02:
The new version of eclipse not yet released as a stable version.

mattb5906, 2017-01-02:
Ah, of course. Thanks Jantje.

jantje, 2017-07-08:
I moved to oxygen for the development environment :-)

jantje, 2017-09-01:
@wimjongman I guess this is done except for #792. So it can be closed.

wimjongman, 2017-09-06:
Close

mjmeijer, 2017-09-06:
mvn clean verify -Pmac64,oxygen fails...
```
Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.updatesite:4.0.1-SNAPSHOT @ /Users/maarten/git/arduino-eclipse-plugin/io.sloeber.updatesite/pom.xml
[INFO] {osgi.os=macosx, osgi.ws=cocoa, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.updatesite raw:4.0.1.'SNAPSHOT'/format(n[.n=0;[.n=0;[-S]]]):4.0.1-SNAPSHOT
[ERROR]   Missing requirement: io.sloeber.updatesite raw:4.0.1.'SNAPSHOT'/format(n[.n=0;[.n=0;[-S]]]):4.0.1-SNAPSHOT requires 'org.eclipse.cdt.feature.group [9.0.0.201606062011]' but it could not be found
```

mjmeijer, 2017-09-06:
In /io.sloeber.updatesite/category.xml remove this for build to work...

`   <feature url="features/org.eclipse.cdt_9.0.0.201606062011.jar" id="org.eclipse.cdt" version="9.0.0.201606062011">
      <category name="Dependencies"/>
   </feature>
`

jantje, 2017-09-07:
[image]

I think it is strange the build succeeds here -I'm not saying you are wrong!!!-
I also dumped my version  platform info and as far as I understand this I'm on 4.7.0.
As far as I remember I tried to upgrade using eclipse upgrade procedure (after having read a article it works fine) and f'ed-up my installation.  So I reinstalled Sloeber. I guess I took a copy from sonatype.

I'm  running mvm clean verify -Pall,oxygen on my sytem and I'll check the eclipse version  it generated.

jantje, 2017-09-07:
As far as I remember
`<feature url="features/org.eclipse.cdt_9.0.0.201606062011.jar" id="org.eclipse.cdt" version="9.0.0.201606062011"> <category name="Dependencies"/> </feature>`
is there to enable the installation of cdt when installing from base eclipse.
Maybe the version number needs to be updated. But this should not impact the product.
On my install cdt is on 9.3

jantje, 2017-09-07:
Strangely enough the local build fails now and the jenkins build works fine :-s
[image]
I'm on master. Only with a additional toolchain in the plugin.xml of io.sloeber.core.

jantje, 2017-09-07:
The local build worked fine and both 64 and 32 bit versions return 4.7
[image]

jantje, 2017-10-08:
given the latest progress I think we can close this issue.

closed 2017-10-08 by jantje

git1234567890, 2017-12-31:
Just to let you know. Installed a fresh Oxygen and added (Marketplace) you plugin.
../Relase/subdir.mk referred ..//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++ ... that folder was not there /arduinoPlugin/packages/arduino/tools!
But with softlinks to the referred programs to the arduino IDE libraries 4.9.2-atmel3.5.4-arduino2 -> /opt/arduino/hardware/tools/avr ALL WORKS FINE!