Sloeber/arduino-eclipse-plugin #577: JVM on Mac

urbhar580, 2016-11-03:
When I try to run the Sloeber app on mac I get a dialog that says:

Version 1.6.0_65 of the JVM is not suitable for this product. Version:1.8 or greater is required. 

I have the latest OS and Java version 8 update 112.

jantje, 2016-11-03:
I don't know anything about mac but it does look to me there is a java 1.6 installed as well and for one reason or another that is the jvm sloeber is using.

wimjongman, 2016-11-08:
Take a look at the sloeber.ini file. Does it contain a vm reference? If not you can add one that points to your java bin directory.

I asked duckduckgo[1] for this issue and there are many references. Here is one [2]

[1] https://duckduckgo.com/?q=Version+1.6.0_65+of+the+JVM+is+not+suitable+for+this+product.+Version%3A1.8+or+greater+is+required.+eclipse&t=ffab&atb=v35-7a_&ia=qa

[2] http://stackoverflow.com/questions/24600860/error-launching-eclipse-4-4-version-1-6-0-65-of-the-jvm-is-not-suitable-for-thi#24703978

jantje, 2016-11-12:
@urbhar580 
Did you fix this issue? Can we close this issue?

urbhar580, 2016-11-20:
I went to Oracles  page for "Java SE Development Kit 8 Downloads" and downloaded the "JDK 8u111". The file name was "jdk-8u111-macosx-x64.dmg". There was also a version 112 but i chose the older one to be safe. The link is below. It is important to chose the JDK (development) and not the JRE (runtime). This solved the problem and Sloeber now starts.

http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

closed 2016-11-20 by urbhar580

jantje, 2016-11-20:
txs for the feedback.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje